lvgl/lvgl #7892: ffmpeg lv_image_set_src issue

AndyEveritt, 2025-03-05:
### LVGL version

v9.2.2

### Platform

VScode with SDL on Linux

### What happened?

When using `lv_image_set_src("assets/ffmpeg.png");` (usage as described [here](https://github.com/lvgl/lvgl/issues/3790) and in the [documentation](https://docs.lvgl.io/9.2/libs/ffmpeg.html)), it fails with the following log messages.
```
[Warn]  (11.843, +11843)         lv_fs_open: Can't open file (assets/ffmpeg.png): unknown driver letter lv_fs.c:88
[Error] (18.063, +6220)  image_decoder_get_info: File open failed: 3 lv_image_decoder.c:332
[Warn]  (27.292, +9229)  lv_image_set_src: failed to get image info: assets/ffmpeg.png lv_image.c:189
```

I believe the error is because in `image_decoder_get_info()` this code is run
```c
    if(src_type == LV_IMAGE_SRC_FILE) {
        lv_fs_res_t fs_res = lv_fs_open(&dsc->file, src, LV_FS_MODE_RD);
        if(fs_res != LV_FS_RES_OK) {
            LV_LOG_ERROR("File open failed: %" LV_PRIu32, (uint32_t)fs_res);
            return NULL;
        }
    }
```
`lv_fs_open()` does not return `LV_FS_RES_OK` since the path does not start with the drv letter, and the function returns before the ffmpeg decoder is ever run.

If I comment out the `return NULL` in the above code snippet, `lv_image_set_src("assets/ffmpeg.png")` does correctly display the image. Although it does print the above error messages at a very rapid rate so those probably need tweaking.

cheungxiongwei, 2025-03-06:
reference https://docs.lvgl.io/master/details/main-modules/fs.html

liamHowatt, 2025-03-06:
Please read https://docs.lvgl.io/master/details/main-modules/fs.html and either set a default driver letter with `LV_FS_DEFAULT_DRIVER_LETTER` or add `A:` (if the driver letter is `A`) to the start of the path. I.e., if your driver letter is `B`, you would have `lv_image_set_src("B:assets/ffmpeg.png");`.

AndyEveritt, 2025-03-06:
@liamHowatt  Ok is that the intended workaround? ffmpeg doesn't use the driver letters by default. 

*from lvgl ffmpeg docs*:
>     FFmpeg extension doesn't use LVGL's file system. You can simply pass the path to the image or video as usual on your operating system or platform.

If this is the intended workaround then the ffmpeg docs & examples should be updated to reflect this imo as the decode image example doesn't work otherwise https://docs.lvgl.io/9.2/libs/ffmpeg.html#decode-image

liamHowatt, 2025-03-06:
Indeed! Let me look into it and I'll open a PR. I forgot that `LV_FFMPEG_PLAYER_USE_LV_FS` existed, and I had it enabled. When disabled, it still tries to use a driver letter. Also `LV_FFMPEG_PLAYER_USE_LV_FS` is not documented.

AndyEveritt, 2025-03-06:
I think `LV_FFMPEG_PLAYER_USE_LV_FS` has been added post v9.2.2 so isn't in a stable release yet

liamHowatt, 2025-03-06:
Wait sorry, leaving `LV_FFMPEG_PLAYER_USE_LV_FS` as `0` does actually have the intended effect in `master`.

So this is a v9.2.2 issue of it always using the LVGL filesystem, contrary to what the docs say? I tested on `release/v9.2` and it works as intended, using the OS path.

When I give an invalid path the error message is different than yours

```
[Error] (0.310, +310)    ffmpeg_open_file: Could not open source file ../lvgl/examples/libs/ffmpeg/birds.mp4 lv_ffmpeg.c:678
[Error] (0.310, +0)      lv_ffmpeg_player_set_src: ffmpeg file open failed: ../lvgl/examples/libs/ffmpeg/birds.mp4 lv_ffmpeg.c:165
[Error] (0.310, +0)      lv_ffmpeg_player_set_cmd: ffmpeg_ctx is NULL lv_ffmpeg.c:214
```

Are you positive you're using a version that doesn't have `LV_FFMPEG_PLAYER_USE_LV_FS`?

liamHowatt, 2025-03-06:
Oh, it's not an issue with MP4, only with image files.

AndyEveritt, 2025-03-06:
I just checked the source code in master and `LV_FFMPEG_PLAYER_USE_LV_FS` is only used when opening video files not image files afaict, so I don't think it is relevant to this issue anyway

liamHowatt, 2025-03-06:
It's also an issue in `master`. It's a hard problem. Loading images with FFMPEG is stuck using the LVGL filesystem unless the LVGL image decoder makes it the responsibility of the decoders themselves to handle opening the files.

I guess just a docs update saying that opening images with FFMPEG always uses the LVGL filesystem?

liamHowatt, 2025-03-12:
Closing as 03c6be3451872ad37e06423ebc67206b7a07da53, part of #7903, has been merged.

PR #7903 (2025-03-12): Do not squash: FFmpeg fixes v9.2
Fixes #7892

This is a copy of #7894 for the `release/v9.2` branch to address the concern in the issue.

The 3rd commit was not a direct cherry-pick. v9.2.2 did not have `LV_FFMPEG_PLAYER_USE_LV_FS` so the docs are different.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_ge [... cut]
closed 2025-03-12 by liamHowatt