lvgl/lvgl #7917: release/v8.4 CI no longer working

liamHowatt, 2025-03-11:
### LVGL version

v8.4

### Platform

`release/v8.4`

### What happened?

https://github.com/lvgl/lvgl/pull/7860#pullrequestreview-2656193115

PRs targeting `release/v8.4`
https://github.com/lvgl/lvgl/pulls?q=is%3Apr+is%3Aopen+base%3Arelease%2Fv8.4

### `check_conf.yml`

```
Run actions/setup-python@v1
  with:
    python-version: 3.7
    architecture: x64
Error: Version 3.7 with arch x64 not found
Available versions:

3.10.16 (x64)
3.11.11 (x64)
3.12.9 (x64)
3.13.2 (x64)
3.8.18 (x64)
3.9.21 (x64)
```

### `ccpp.yml`

```
#3 [internal] load metadata for docker.io/arm64v8/debian:bullseye
#3 ERROR: no match for platform in manifest: not found
------
 > [internal] load metadata for docker.io/arm64v8/debian:bullseye:
------
Dockerfile.aarch64.bullseye:1
--------------------
   1 | >>> FROM arm64v8/debian:bullseye
   2 |     
   3 |     COPY ./run-on-arch-install.sh /root/run-on-arch-install.sh
--------------------
ERROR: failed to solve: arm64v8/debian:bullseye: failed to resolve source metadata for docker.io/arm64v8/debian:bullseye: no match for platform in manifest: not found
Error: The process '/home/runner/work/_actions/uraimo/run-on-arch-action/v2.1.1/src/run-on-arch.sh' failed with exit code 1
```

kisvegabor, 2025-03-25:
Can you take a look in the not too distant future?

kisvegabor, 2025-05-20:
Both are merged, but still failing. 🙁

closed 2025-07-06 by lvgl-bot