lvgl/lvgl #7918: lv_display_set_rotation does not work correctly while using ST LTDC driver

userganeshr, 2025-03-11:
### Introduce the problem

Hi,
    By following this [link](https://github.com/lvgl/lvgl/issues/7178) while try to rotate the screen 90 degree by using stm32 ltdc and lvgl v9.3.0. I have created display using lvgl partial method like below, but the rendered screen looks not correct, i have attached the image below. anyone please guide me to solve this issue.

attribute((section(".sdram_data"), aligned(32))) static lv_color_t buf_1[800 * 480];
attribute((section(".sdram_data"), aligned(32))) static lv_color_t buf_2[800 * 480];

lv_display_t * disp = lv_st_ltdc_create_partial(buf_2, NULL, sizeof(buf_1), 0);
lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90); /* set landscape orientation */

[image]

liamHowatt, 2025-03-11:
Hmm what does `180` and `270` look like? I assume `0` rotation is ok? Are you using `LV_USE_OS`?

Some notes
- `buf_1` is unused
- What is the color depth of the display? Declaring your buffers as arrays of `lv_color_t` means arrays of 3 byte structs, which can make sense for 24 bit color depth i.e. RGB888, but if the display is 32 bit, i.e. ARGB8888 then the buffers may be smaller than you intended. But it'd be ok since partial rendering mode works with smaller buffers.

userganeshr, 2025-03-12:
Hi @liamHowatt ,
     Thanks for the response. I am not using LV_USE_OS. Now, i have changed my lvgl partial creation like below by using both buffers.
  **lv_display_t * disp = lv_st_ltdc_create_partial(buf_1, buf_2, sizeof(buf_1), 0);
  lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);     /* set landscape orientation */**

And color depth was 16bit, i am using RGB565. In below i have shared my ltdc configuration also. After using both buffer for rendering, the screen distortion looks slightly better. but not completely resolved. In below i am sharing the current status of the screen. There is one line of distortion at edge of the display. can you please help me to resolve this ??

[image] 

[image]

closed 2025-04-08 by lvgl-bot