lvgl/lvgl #7921: Lottie with webp,png embedded not rendering

CodingPapi, 2025-03-12:
### LVGL version

V9

### Platform

all

### What happened?

Dear all,
currently render Lottie with ThorVG show blank view if lottie files contains png or webp embedded data encoded with base64.
ThorVG has loders for png and webp data, it seems that lvgl not contains those loaders.
Could you pls check this?
you can use attached cat.json to reproduce

### How to reproduce?

[cat.json](https://github.com/user-attachments/files/19202320/cat.json)

CodingPapi, 2025-03-14:
Can we just support one of them? only PNG or webp, cause many lotties are converted from other formats, or designed with pics

liamHowatt, 2025-03-14:
I think it could be reasonable. Do you know what it would take to add it?

closed 2025-04-08 by lvgl-bot