lvgl/lvgl #7923: docs on release life cycle

faxe1008, 2025-03-12:
### Introduce the problem

It's not entirely clear how long a given release is supported. Meaning the policy on backporting fixes and possible security vulns. A central page documenting each release and its support state would be very nice. Also anticipated dates for releases etc.

Also in that context, what would be extremely handy is to have a label maintainers can attach to a PR like "backport-<release>" which triggers creation of a PR targeting the release branch with the fix. Because I feel like not every fix is backported to its respective release.

### Proposal

As outlined a dedicated page would be neat :^)

AndreCostaaa, 2025-03-12:
I believe you're looking for: https://docs.lvgl.io/master/intro/introduction.html#version-support

faxe1008, 2025-03-15:
Ah indeed. Hmm seems to be a SEO or on page search issue that I was not able to find it. Thanks 👍

closed 2025-03-15 by faxe1008