lvgl/lvgl #7930: Can't work with PNG Raw Data

ats3788, 2025-03-13:
### LVGL version

v9.2.2

### Platform

PlatformIo ESP32 I can't display raw PNG Images. What is odd I have an Test Project and it works fine with an PNG Image but if I do the same in a new Project the Image is just not displayed. Even I switch on LV_LOG.......
I don't get any Error or Warnings, the Image will not be shown. Any clue what going on.

### What happened?

To display the Image

ats3788, 2025-03-13:
H3ello what is even more odd
If I use LV_conf.h Configuration file for v9.0.0 The Raw Data will be displayed
If I use Configuration file for v9.2.2 I can Load PNGs from an SD Card but not the raw Image in an Array, what do I miss.
In the Zip File I have attached an Ver 9.0.0 and Ver 9.2.2 lv_conf.h

[lv_conf.zip](https://github.com/user-attachments/files/19227927/lv_conf.zip)

liamHowatt, 2025-03-14:
Looks like `LV_USE_LODEPNG` is not enabled in the `lv_conf.h.bak`. Enable it and let me know if it works.

ats3788, 2025-03-19:
What is so super ODD I run is with lv_conf.h Ver 9.0.0 the PNG will be displayed but does not work with png and SC Card. Now I use my old Ver v9.2.2 and the png will be displayed Super Super OD

closed 2025-04-13 by lvgl-bot