lvgl/lvgl #7931: Non-transparent background when using a line

iSetebos, 2025-03-13:
### LVGL version

v9.2

### Platform

```
 static lv_style_t style_line;
    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, 6);
    lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_rounded(&style_line, true);
    lv_style_set_bg_opa(&style_line, LV_OPA_TRANSP); // Set background opacity to transparent

       lv_obj_t *obj = lv_line_create(Panel); //
            static lv_point_precise_t line_points[] = {
                { 0, 0 },
                { 50, 50 },
                { 100, 0 },
                { 150, 50 },
                { 200, 0 }
            };

            lv_line_set_points(obj, line_points, 5);
             lv_obj_add_style(obj, &style_line, 0);
            lv_obj_set_style_bg_opa(obj, 0, 0);
            lv_obj_set_size(obj, LV_SIZE_CONTENT, LV_SIZE_CONTENT);
             lv_obj_center(obj);
```

[image]

### What happened?

When using the widget, the background is not transparent for some reason.

liamHowatt, 2025-03-14:
Hmm I cannot reproduce the issue. Is there anything special or different about your setup?

iSetebos, 2025-03-14:
Init code
`    disp = lv_display_create(TFT_HOR_RES, TFT_VER_RES);
    lv_display_set_flush_cb(disp, my_disp_flush);
    lv_display_set_buffers(disp, draw_buf, NULL, sizeof(draw_buf), LV_DISPLAY_RENDER_MODE_PARTIAL);
    lv_display_set_color_format(disp, LV_COLOR_FORMAT_ARGB8888); 

    ui_osd = lv_obj_create(NULL);
    lv_obj_clear_flag(ui_osd, LV_OBJ_FLAG_SCROLLABLE | LV_OBJ_FLAG_SCROLL_ELASTIC | LV_OBJ_FLAG_SCROLL_MOMENTUM |
                      LV_OBJ_FLAG_SCROLL_CHAIN);
    lv_disp_load_scr(ui_osd); 
    // - Enable transparent background
    lv_obj_set_style_bg_opa(ui_osd, LV_OPA_TRANSP, LV_PART_MAIN);
`