lvgl/lvgl #7936: Missing headers when enabling LV_FREETYPE_USE_LVGL_PORT

friedkeenan, 2025-03-15:
### LVGL version

v9.2.0-569-g8c313b230

### Platform

SDL on Linux, and Nintendo Switch homebrew

### What happened?

I want to use Freetype to render fonts from a buffer in memory, so after some investigating I figured out that I could use LVGL's "memfs" interface to hand off that buffer to LVGL's Freetype API, which expects file paths. And judging by [this warning](https://github.com/lvgl/lvgl/blob/62857c4b713eebd424a5388212b7f07c074f18cb/src/libs/freetype/lv_freetype.c#L318), it would seem that I also need to enable `LV_FREETYPE_USE_LVGL_PORT` so that Freetype will use LVGL's filesystem APIs, enabling it to read the memfs file.

However, when I do that, I get a compiler error upon compiling `lv_ftsystem.c` when it [tries to include internal Freetype headers](https://github.com/lvgl/lvgl/blob/62857c4b713eebd424a5388212b7f07c074f18cb/src/libs/freetype/lv_ftsystem.c#L15-L16) which, as far as I can tell, simply do not exist on my system.

The only explanation for this that I can figure out is either that I'm expected to *only* use Freetype as a submodule or similar, which would end up having those internal headers still present, or that I'm expected to just nab those internal headers from Freetype and call it a day? Or that I'm just missing something obvious. After countless tabs and lots of searching, it would appear like I'm the only one with this issue, so that could definitely be the case. But I'd like to just make sure of what I'm supposed to be doing, and for that to eventually be searchable for others as well.

I'd also like to say that I really hope that integrating third party dependencies with LVGL when building can be sorted out. As someone who's using meson as my build system, which is interfacing with LVGL's CMake file to use it as a library, it has been pretty difficult conveying Freetype to LVGL's build process. That's not core to this issue, but I thought I should mention it because there were a lot of other headers that LVGL could not initially find either, because it didn't know where to look for Freetype's headers (which I'm still not entirely certain the best way to solve, but I do know that I *can* solve it). But I saw from #7223 that work is being done on that, so take this as a word of support towards that I guess.

### How to reproduce?

Enable both `LV_USE_FREETYPE` and `LV_FREETYPE_USE_LVGL_PORT`, and tell LVGL where to find Freetype's headers, and then compile LVGL.

friedkeenan, 2025-03-17:
Ok, so I'm trying to just get something that at least works while waiting for any info on this, so right now I'm just nabbing the internal headers from Freetype, but when I tried to compile I got an error because `FT_ERR_PREFIX` was undefined, it would seem, based on [this code from Freetype](https://github.com/freetype/freetype/blob/5d4e649f740c675426fbe4cdaffc53ee2a4cb954/include/freetype/fterrors.h#L232) that it's needed to define `FT2_BUILD_LIBRARY` to keep `FT_ERR_PREFIX` defined. Defining that before including the Freetype headers in `lv_ftsystem.c` does then let LVGL compile.

However, upon trying to actually use the memfs interface with the Freetype interface, I run into an issue that.. I cannot actually make a memfs path. The [function to make one](https://github.com/lvgl/lvgl/blob/1247f77eb1c563428851ff7be47d6b4b0015c08a/src/misc/lv_fs.h#L151) is public, however [the body of the actual `lv_fs_path_ex_t` type](https://github.com/lvgl/lvgl/blob/1247f77eb1c563428851ff7be47d6b4b0015c08a/src/misc/lv_fs_private.h#L35-L39) is private, seemingly ever since #6068. LVGL's Freetype interface doesn't have a `create_from_buffer` function like the binfont interface does, so I'm really unsure how I'm meant to pass it the TTF buffer. At this point I feel like I'm possibly the only one who has ever tried to do this, at least in quite a long time.

closed 2025-04-13 by lvgl-bot

b272488049, 2025-04-21:
Same here: translation

[image]