lvgl/lvgl #7938: LVGL Zephyr support

kisvegabor, 2025-03-17:
### Introduce the problem

[Here](https://forum.lvgl.io/t/nemagfx-support-for-vg/20346/3?u=kisvegabor) it came up how to support G2D rendering in Zephyr. It's in LVGL but it uses `lv_thread_` (LVGL's OS API in general).

Ideally the users just need to enable `LV_USE_DRAW_G2D` and `LV_USE_OS LV_OS_ZEPHYR` but we don't have `LV_OS_ZEPHYR` in LVGL.

### Proposal

1. Shall we add `LV_OS_ZEPHYR` in LVGL?
2. Is it added in Zephyr? If so we need to document it somewhere.

cc @faxe1008 @uLipe @anaGrad @cosmindanielradu19

liamHowatt, 2025-03-17:
Zephyr supports pthreads, FYI. I don't know anything else about it though.

uLipe, 2025-03-17:
We need also to consider that the hardware underlying under G2D implementation may need to map correctly its interrupts, so we need to take that into account because zephyr manages the soc interrupts internally

uLipe, 2025-03-17:
To add a bit more of context to my answer above, a friend of mine also wanted to use hardware acceleration but in Nuttx, I was talking to him and after figuring out in the code I noticed that the HW acceleration peripherals port are strongly related to the SoC Vendor SDK.

In general more complex GPU have two major components:

Kernel Driver -> part of the target platform it is responsible to initialize the peripheral, maps the interrupts, and memory, and finally makes it proper powered for usage, in general the kernel driver needs to expose a minimal peripheral hal which is the bridge to the let's say LVGL side.

LVGL GPU Client API -> part of the LVGL side, in general the client API follows a standard (like VGLite or OpenGL), and consumes a set of hal API exposed by the kernel driver, in general the only hardware relevant step here is to proper sync the LVGL initialization order with the peripheral / platform initialization steps, once the platform initializes the peripheral and as long the hal functions expected for the client API are implemented on the target platform, the client API is exposed to the rest of the LVGL as one of its currently enabled draw units.

For simple HW acceleration peripherals, let's say DMA2D, PxP, and even D/AVE, the kernel driver set of functions expected from the LVGL relevant client API should be at least: the OS Absttraction layer (already covered by Zephyr),  and a minimal set of MMIO operations and an abstraction of the platform interrupt connecting APIs, I think this can make the draw units usable regardless the target platform and defer the peripheral specifiic things to the layer that they belong vendor HAL/SDK.

kisvegabor, 2025-03-18:
> There is a Kconfig symbol you can enable LV_Z_USE_OSAL, which pulls in our OSAL implementation.

Great! Would it make sense to add it to LVGL?

> the OS Absttraction layer (already covered by Zephyr), and a minimal set of MMIO operations and an abstraction of the platform interrupt connecting APIs, I think this can make the draw units usable [...]

- **Operating System APIs (already covered):** Clear
- **MMIO minimal set of API per Platform / OS**: My approach on that IO inits should be done outside of LVGL. If LVGL needs to manage IOs, we should use callbacks. I feel like abstracting IOs and peripheries in general would be a huge task, and it's already handles by SDKs or OSes like Zephyr or NuttX. It already worked well [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/display/lcd/lv_lcd_generic_mipi.h#L186).
- **Interrput managment API per Platform / OS** I don't have a good idea on that, and we already faced this issue e.g. in case of DMA2D. As interrupts can be vendor, OS, and compiler specific I feel like a callbacks work the best here too. 

As an example our DMA2D used the the periphery on [registrer level](https://github.com/lvgl/lvgl/blob/master/src/draw/dma2d/lv_draw_dma2d.c#L161) but for this we need a custom [include](https://github.com/lvgl/lvgl/blob/master/src/draw/dma2d/lv_draw_dma2d_private.h#L22C10-L22C35). So it requires an STM32 header. Does it work in Zephyr? How does Zephyr gets the register layout?

liamHowatt, 2025-03-18:
DMA2D also requires outside code to call `lv_draw_dma2d_transfer_complete_interrupt_handler` on interrupt.

From https://docs.lvgl.io/master/details/integration/chip/stm32.html#dma2d-support:
> If `LV_USE_DRAW_DMA2D_INTERRUPT` is enabled then you are required to call `lv_draw_dma2d_transfer_complete_interrupt_handler` whenever the DMA2D "transfer complete" global interrupt is received.

The reason it's done this way is because STM32 has a **global** interrupt handler. The STM32 HAL can decode the interrupt cause call a more specific handler but the LVGL DMA2D implementation actually doesn't depend on the HAL. For maximum flexibility/modularity, This LVGL driver does not presume handle the global interrupt on its own, nor depend on any HAL or RTOS interface. The only downside is the driver user is burdened with the task of arranging to call this function.

kisvegabor, 2025-03-23:
I think it's a good idea! We can have `draw_unit->tasks_completed_cb()` and a `lv_draw_unit_tasks_completed(draw_unit)` function. It'd be a sweet spot between abstractions and plug-and-play setup.

kisvegabor, 2025-03-23:
@faxe1008  What about this?

>> There is a Kconfig symbol you can enable LV_Z_USE_OSAL, which pulls in our OSAL implementation.

> Great! Would it make sense to add it to LVGL?

I assume that Zephyr's OS API is quite stable so it doesn't require much maintanence.

kisvegabor, 2025-04-07:
@faxe1008 Can you comment on [this](https://github.com/lvgl/lvgl/issues/7938#issuecomment-2746163992)?

faxe1008, 2025-04-09:
Morning @kisvegabor,

sorry for the late reply kinda forgot about this one. The OS part is mostly stable (semaphore, mutexes) but especially the dynamic thread creation part is not yet:
https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_DYNAMIC_THREAD

Additionally we have some glue code that needs to live within Zephyr itsself, like the device bindings for the input pseudo devices. I do not think it would be a good idea to have those within LVGL, since they are extremely specific to how Zephyr works and also subject to changes. Since there are some parts that have to live with Zephyr its kinda unwieldy to have parts eg. the OSAL live upstream. 

There is quite a history of the code moving back and forth, see the latest discussion:
https://github.com/zephyrproject-rtos/zephyr/issues/60868

If there are no major concerns with the locations where the code resides atm. I am inclined to leave things as is.

uLipe, 2025-04-11:
I would go to the same direction, the platform side code for Zephyr should live there, and maintaned on the kernel side.

Just the the current OSAL to live in LVGL would keep the porting of the LVGL smooth as is it currently.

The specifics regarding GPUs or draw units are another thing, they need to have the generic code on the LVGL and provide the porting layer this one to be implemented on tbe platform code side, as is done currently on VG_Lite

kisvegabor, 2025-04-15:
You know it better than me what is the best for the maintenance of Zephyr, so I'm ok with keep it in Zephyr. 

I think we should mention it in our [Zephyr docs](https://docs.lvgl.io/master/details/integration/os/zephyr.html).

markgrebe, 2025-04-16:
Has anyone successfully used Zephyr 4.1 and LVGL 9 with DMA2D and interrupts?  We have tried on the Riverdi board and the stm32u5g9j_dk2 board, and have not been able to get it to work.

If we use no interrupts:
CONFIG_LV_USE_DRAW_DMA2D_INTERRUPT=n
CONFIG_LV_Z_USE_OSAL=n

We can see the DMA2D successfully complete.  But if we set both of those to yes, and hook up interrupts as so:

````
void dma2d_isr(void *arg)
    {
    lv_draw_dma2d_transfer_complete_interrupt_handler();
    }

void init_dma2d()
    {
    IRQ_CONNECT(DMA2D_IRQn, 0, dma2d_isr, NULL, 0);
    irq_enable(DMA2D_IRQn);
    }

```
calling init_dma2d() from main(), (We do plan on doing a proper driver/dts, but we are just trying to make sure this will work at this point).

We never make it to the interrupt handler.  We have tried enabling all the DMA2D interrupt bits, thinking we may have been getting an error, but no joy there either.

markgrebe, 2025-04-28:
@uLipe thanks so much for the reply, it turns out my issues were pilot error on my part.  I was having issues with interrupts on the 8.x version of the driver I was working on  for another customer, and I made the wrong assumption that I had the same issues on 9.x.  I didn't realize that LVGL was starting up 2 drawing threads when you enable the OSAL (one SW and one DMA2D for my config), and the SW stack size was greater than what I had allocated for my dynamic thread pool.  So I wasn't even getting to where the interrupts were needed.

@liamHowatt, I did find 3 issues with the driver that I thought I would let you know about when I got DMA2D working with Zephyr.   I can also do a pull request if that would be easier.

1.  I needed to add src/osal/lv_os.h to the main lvgl.h include.  (That may be Zephyr specific).
2. The DMA2D interrupt flags need to be cleared in the Interrupt Service Routine.  This could be left for the OS side routine, but since all OS's are going to require this, and the driver does all the other register writing, I did this before returning in lv_draw_dma2d_transfer_complete_interrupt_handler():

`DMA2D->IFCR = 0x3FU; // trigger ISR flags reset`

Without that, you end up in an infinite interrupt loop, reentering the ISR as soon as you exit.
 
3.  In thread_cb() in lv_draw_dma2d.c, the do/while loop proved to be problematic.  The issue is that in it's test. the field u->task_act is only set to NULL after the loop in post_transfer_tasks(), so the loop ends up being infinite.  I replaced:

`do {
       lv_thread_sync_wait(&u->interrupt_signal);
} while(u->task_act != NULL);`

with:

` lv_thread_sync_wait(&u->interrupt_signal);`

and everything worked great.

One last comment, it should probably be noted in the documentation somewhere that the LVGL configuration item LV_ATTRIBUTE_MEM_ALIGN_SIZE should be set to 4 (if you plan on using any 32 bit color formats like ARGB8888 or XRGB8888, since DMA2d expects them to be 32 bit aligned.  I got caught by that chasing why the images in the Music and Widget demos weren't working.

faxe1008, 2025-04-29:
@markgrebe Curious normally when you enable the zephyr OSAL `LV_USE_PRIVATE_API` should be pulled in.
See:
https://github.com/zephyrproject-rtos/zephyr/blob/4d848ebdb1148794189661b7e5741c0f96ee22b8/modules/lvgl/Kconfig#L165
Can you check your `autoconf.h` if that is the case?

markgrebe, 2025-05-02:
@faxe1008 sorry for the delay in responding, I got tied up on another project, and just got back to this.

I did check that, and you are correct, LV_USE_PRIVATE_API did get pulled in.  In fact, I went back to my dma2d change and took the change out of the header, and everything works.  It must have been an artifact of something I was doing before I had the config correct, so you can ignore my comment about the header, sorry for the false positive.

closed 2025-05-26 by lvgl-bot