lvgl/lvgl #7942: Maybe someone can explain why this is happening.

kdschlosser, 2025-03-17:
### LVGL version

9.2.2

### Platform

ESP32

### What happened?

For whatever reason it is taking longer to render to a smaller frame buffer. I am not sure what would cause this...

```
#define BUFFER_SIZE (DISP_HOR_RES * DISP_VER_RES * sizeof(uint16_t) / 10)
```
[image]

```
#define BUFFER_SIZE (DISP_HOR_RES * DISP_VER_RES * sizeof(uint16_t) / 8)
```
[image]

The only difference in between the 2 videos is the `BUFFER_SIZE`. Why is it taking longer for the smaller buffer to render?

### How to reproduce?

```c
#include "esp_log.h"

#include <stdio.h>
#include <unistd.h>
#include <sys/lock.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_st7789.h"
#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "esp_check.h"
#include "esp_err.h"
#include "esp_log.h"
#include "lvgl.h"

static const char *TAG = "MyDisplay";

/* LCD size */
#define DISP_HOR_RES 320
#define DISP_VER_RES 172

/* LCD settings */
#define DISP_DRAW_BUFF_HEIGHT 50

/* LCD pins */
#define DISP_SPI_NUM SPI2_HOST
#define DISP_GPIO_SCLK GPIO_NUM_12
#define DISP_GPIO_MOSI GPIO_NUM_11
#define DISP_GPIO_RST -1 // Not connected
#define DISP_GPIO_DC GPIO_NUM_47
#define DISP_GPIO_CS GPIO_NUM_45
#define DISP_GPIO_BL GPIO_NUM_48

#define BUFFER_SIZE (DISP_HOR_RES * DISP_VER_RES * sizeof(uint16_t) / 10)

static esp_lcd_panel_handle_t panel_handle = NULL;
static esp_lcd_panel_io_handle_t io_handle = NULL;
static lv_display_t *display = NULL;

static void *buf1 = NULL;
static void *buf2 = NULL;

/* Function prototypes */

lv_obj_t *lv_example_slider_1(void);

// This gets called when the DMA transfer of the buffer data has completed
static bool notify_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
{
    lv_display_t *disp = (lv_display_t *)user_ctx;
    lv_display_flush_ready(disp); // I have tried to change this to the global display variable, no change
    return false;
}

static void flush_cb(lv_display_t *disp, const lv_area_t *area, uint8_t *px_map)
{
    int x1 = area->x1;
    int x2 = area->x2;
    int y1 = area->y1 + 34; // Offset image to compensate for smaller 172px resolution
    int y2 = area->y2 + 34; // Offset image to compensate for smaller 172px resolution

    // uncomment the following line if the colors are wrong
    lv_draw_sw_rgb565_swap(px_map, (x2 + 1 - x1) * (y2 + 1 - y1)); // I have tried with and without this

    esp_lcd_panel_draw_bitmap((esp_lcd_panel_handle_t)lv_display_get_user_data(disp), x1, y1, x2 + 1, y2 + 1, px_map);
}

static esp_err_t lvgl_init(void)
{
    display = lv_display_create(DISP_HOR_RES, DISP_VER_RES);

    buf1 = heap_caps_calloc(1, BUFFER_SIZE, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA);
    buf2 = heap_caps_calloc(1, BUFFER_SIZE, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA);

    lv_display_set_buffers(display, buf1, buf2, BUFFER_SIZE, LV_DISPLAY_RENDER_MODE_PARTIAL);

    lv_display_set_user_data(display, panel_handle);
    lv_display_set_color_format(display, LV_COLOR_FORMAT_RGB565);

    lv_display_set_flush_cb(display, flush_cb);

    const esp_lcd_panel_io_callbacks_t cbs = {
        .on_color_trans_done = notify_flush_ready,
    };
    /* Register done callback */
    ESP_RETURN_ON_ERROR(esp_lcd_panel_io_register_event_callbacks(io_handle, &cbs, display), TAG, "esp_lcd_panel_io_register_event_callbacks error"); // I have tried to use
    ESP_RETURN_ON_ERROR(esp_lcd_panel_init(panel_handle), TAG, "esp_lcd_panel_init error");

    // Hardware rotate 90°
    uint8_t madctl = 0x60; // Rotate reg 0x00 0x60 0xC0 0xA0
    esp_lcd_panel_io_tx_param(io_handle, 0x36, &madctl, 1);

    esp_lcd_panel_io_tx_param(io_handle, 0x21, NULL, 0); // Inverted color fix (to get normal colors)

    return ESP_OK;
}

static esp_err_t display_init(void)
{
    // LCD backlight and DC
    gpio_config_t gpio_cfg = {
        .pin_bit_mask = (1ULL << DISP_GPIO_BL) | (1ULL << DISP_GPIO_DC),
        .mode = GPIO_MODE_OUTPUT,
    };
    ESP_ERROR_CHECK(gpio_config(&gpio_cfg));

    gpio_set_level(DISP_GPIO_BL, 1); // Turn on backlight
    gpio_set_level(DISP_GPIO_DC, 1); // Default to data mode

    // LCD initialization
    ESP_LOGD(TAG, "Initialize SPI bus");
    spi_bus_config_t buscfg = {};
    buscfg.sclk_io_num = DISP_GPIO_SCLK;
    buscfg.mosi_io_num = DISP_GPIO_MOSI;
    buscfg.miso_io_num = GPIO_NUM_NC;
    buscfg.quadwp_io_num = GPIO_NUM_NC;
    buscfg.quadhd_io_num = GPIO_NUM_NC;
    buscfg.max_transfer_sz = BUFFER_SIZE; // DISP_HOR_RES * DISP_DRAW_BUFF_HEIGHT * sizeof(uint16_t);
    ESP_RETURN_ON_ERROR(spi_bus_initialize(DISP_SPI_NUM, &buscfg, SPI_DMA_CH_AUTO), TAG, "SPI init failed");

    esp_lcd_panel_io_spi_config_t io_config = {
        .dc_gpio_num = DISP_GPIO_DC,
        .cs_gpio_num = DISP_GPIO_CS,
        .pclk_hz = 80 * 1000 * 1000,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .spi_mode = 0,
        .trans_queue_depth = 10,
    };
    ESP_RETURN_ON_ERROR(esp_lcd_new_panel_io_spi(DISP_SPI_NUM, &io_config, &io_handle), TAG, "SPI init failed");

    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = DISP_GPIO_RST,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,
        .bits_per_pixel = 16,
        .flags = {.reset_active_high = 0}};

    ESP_RETURN_ON_ERROR(esp_lcd_new_panel_st7789(io_handle, &panel_config, &panel_handle), TAG, "Display init failed");

    // Reset the display
    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));

    // Initialize LCD panel
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));

    // Vendor specific settings

    // Turn on the screen
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));

    return ESP_OK;
}

static void lvgl_tick_increment(void *arg)
{
    // Tell LVGL how many milliseconds have elapsed
    lv_tick_inc(2);
}

static esp_err_t lvgl_tick_init(void)
{
    esp_timer_handle_t tick_timer;

    // Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = &lvgl_tick_increment,
        .name = "LVGL tick",
    };
    ESP_RETURN_ON_ERROR(esp_timer_create(&lvgl_tick_timer_args, &tick_timer), TAG, "Creating LVGL timer filed!");
    return esp_timer_start_periodic(tick_timer, 2 * 1000); // 2 ms
}

static void lvgl_task(void *arg)
{

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    esp_log_level_set("lcd_panel", ESP_LOG_VERBOSE);
    esp_log_level_set("lcd_panel.st7789", ESP_LOG_VERBOSE);
    esp_log_level_set(TAG, ESP_LOG_VERBOSE);

    esp_err_t ret = display_init();

    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "ST7789 failed to initilize");
        while (1)
            ;
    }
    ret = lvgl_init();
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "LVGL Display failed to initialize");
        while (1)
            ;
    }

    // Create a slider
    lv_obj_t *slider = lv_example_slider_1();

    // Handle LVGL tasks
    int32_t count = 1;
    int32_t increment = 1;
    lv_slider_set_value(slider, 1, LV_ANIM_ON);
    lv_refr_now();

    while (1)
    {
        vTaskDelay(pdMS_TO_TICKS(10));
        if (count == 100 || count == 0) {
            increment = -increment;
        }
        count += increment;
        lv_slider_set_value(slider, count, LV_ANIM_ON);
        lv_refr_now(NULL);

        lv_task_handler();
    }
}

lv_obj_t *lv_example_slider_1(void)
{
    /*Create a screen*/
    lv_obj_t *screen = lv_screen_active();
    lv_obj_set_style_bg_color(screen, lv_color_black(), 0);

    /*Create a slider in the center of the display*/
    lv_obj_t *slider = lv_slider_create(lv_screen_active());
    lv_obj_center(slider);
    lv_obj_set_height(slider, 30);

    /* Change indicator (filled part) color */
    lv_obj_set_style_bg_color(slider, lv_palette_main(LV_PALETTE_GREEN), LV_PART_INDICATOR);
    lv_obj_set_style_bg_color(slider, lv_palette_main(LV_PALETTE_LIME), LV_PART_KNOB);

    return slider;

}

// we are going to have the tick run on core 0 This is so the interrupt is
// also executed on core 0. I move the lvgl task to core 1 and bumped up
// the priority to 22 so it will have the highest priority.

void tick_task(void *args)
{
    ret = lvgl_tick_init();
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Timer failed to initialize");
        while (1)
            ;
    }
    vTaskDelete(NULL);
}

void app_main()
{

    vTaskDelay(1000 / portTICK_PERIOD_MS);

    TaskHandle_t taskHandle = NULL;
    TaskHandle_t tickHandle = NULL;

    lv_init();

    xTaskCreatePinnedToCore(lvgl_task, "LVGL task", 8192, NULL, 22, &taskHandle, 1); // stack, params, prio, handle, core
    xTaskCreatePinnedToCore(tick_task, "LVGL tick task", 4096, NULL, 4, &tickHandle, 0); // stack, params, prio, handle, core

    ESP_LOGI("CoreCheck", "Running on core %d", xPortGetCoreID());
}
```

kdschlosser, 2025-03-17:
I don't believe the time measurement is captured per "frame" it is captured per buffer rendered.. and a "frame" update is not always going to use 10 buffers to update the frame. So that logic doesn't apply.

kdschlosser, 2025-03-17:
the reason why I say that is because they both have the same 50fps. and that would be impossible with one showing an average render time of 11ms and the other with a 14ms.  if the render time was referring to the amount of time it takes to render a "frame" update which may or may not be multiple partial buffers being rendered with those average times you would be seeing a frame rate of 90fps and 71fps and not 50fps.

liamHowatt, 2025-03-18:
Oh, right, it doesn't update the whole screen, but it may need to use more buffers to finish a render round, though. The 1/8th buffer will be big enough to render some parts fewer times than the 1/10th buffer, like the slider indicator. You may see with `LV_USE_REFR_DEBUG`.

The FPS is probably capped at 50 because you have set `LV_DEF_REFR_PERIOD` to `20`.

kdschlosser, 2025-03-18:
what I am saying is the FPS is tied to the "frame" update I believe. A frame update could be one or more partial buffers. The time that is seen in the system monitor consists of 2 numbers. they are in the parenthesis after the time. those 2 numbers are the time it takes to render a single partial buffer and the time it takes to flush the buffer. You can see the flush times are really low. That's because of DMA memory use. that first number however is a single buffer render time. and when the buffer is smaller it takes more time to render. As a guess I would say it's because of LVGL needing to calculate what needs to be rendered into that specific buffer. That is a fairly large amount of time that is getting consumed by making those calculations. on average it's a 3 millisecond delay and that is with only a single widget being involved. You start adding more widgets and having multiple widgets update during a single frame that time is going to grow. This is probably why I have seen render times taking ~40 milliseconds without rendering gradients or widgets.

Maybe the code needs to be looked over and some optimizations to be performed.  If the performance can be increased using a small amount of additional memory it may be worth the tradeoff.

In the specific test I ran I know that LVGL  was able to render the update to a single buffer with one test where it required 2 buffers for the second test. If the smaller buffer is taking 14 milliseconds and 2 are needed to update the frame you end up with a total render time of 28 milliseconds where as the larger buffer only takes 11 milliseconds. It really doesn't seem like it would be worth it to run a smaller buffer unless there is a memory constraint. I know there would be some additional over head but you are talking about a 27% impact on a display that is 172 x 320 x 16bpp with the buffer size being 11,008 for the small partial buffers and 13,760 bytes for the larger partial buffers.

This also shows there is a serious calculation error in the performance monitor because there is no way they would both have identical frame rates if one is taking 28ms to render while the other is taking 11ms. `1000 / 28 = 35` and `1000 / 11 = 90`. Where exactly is the 50 coming from?

kdschlosser, 2025-03-18:
There is a regression because when I set the buffer size even larger to 17,200 the performance gained ends up being lost.

Knowing this I am thinking that some code can be added to LVGL to dynamically adjust the needed frame buffer size to optimize performance. what I mean by that is if I set LVGL tom render to a frame buffer that is 17,200 but the area that needs to be updated would only fill 1/2 of that buffer then LVGL should be only filling 1/2 of it. This is not the case. LVGL will fill the entire buffer rendering areas that don't need to be rendered. It should not be hard to implement this kind of a feature and for devices that are not memory constrained a user could supply 2 buffers that are the full frame is size and LVGL would dynamically shrink what is being rendered to only the areas that are actually needing to be updated so you end up getting rid of that added time for rendering areas that don't need to be and you don't end up adding time because of needing to split an update across multiple partial buffers.

kdschlosser, 2025-03-18:
The calculation do determine if an update would be faster splitting it into 2 partials or rendering all the updates at once should be fairly easy to figure out. That calculation would be based on the amount of area that is getting rendered but nothing has actually updated VS the amount of area that is getting rendered and actually needs to be rendered. Those numbers bounced off the total buffer size and you should be able to come up with the best way to perform the update

I like pictured to describe things it easier that way..

you have a UI where there are 2 areas that need to be updated. Those areas are next to each other. That is what the red boxes are. The blue box is what LVGL would render in a single shot only using a small amount of the supplied frame buffers.

[image]

But if the areas are on opposite ends on the display instead of LVGL rendering this massive area where no updated have taken place
[image]

It makes the decision to render 2 times and the amount of rendering has been reduced by a significant amount.

[image]

Because there is a regression in the performance if the buffer is too large in size for the update at hand that means there is a threshold on either end. If the buffer is too small there is a performance hit and if the buffer is too large there is a performance hit. Making LVGL "smart" to make a decision on what is the optimal way to render the data would be a fantastic thing to have. I do not believe that the cost for doing that calculation would be that much seeing as how LVGL already iterates over the objects to see which ones need updating. collecting that area that needs to be updated at the same time would not add much in the way of overhead. Calculating the bounding rectangle for all of the areas that need to be updated and adding all of the total areas together and getting the ratio between the 2 will tell LVGL how much area that doesn't have anything that has updated would need to be rendered. If it crosses a specific threshold then it splits the rendering into chunks that don't eclipse the hard coded ratio limit.

liamHowatt, 2025-03-18:
It already does what you show in the final picure.

kdschlosser, 2025-03-18:
No it doesn't.

Lets say I am using a display that is 480 x 320 x 16bpp.
If I pass a frame buffer to LVGL that is 480 x 320 x 2 in size and I set the render mode to partial what does LVGL do when it renders? It will render the entire screen. It doesn't just single out the small areas that need to be rendered and pass a truncated frame buffer to the flush callback. It renders EVERYTHING and passes the entire frame buffer to the flush callback. 

If LVGL dynamically reduced the size of the rendered area to only encompass all areas that have been updated omitting the areas that have no updates the rendering would be faster? Yes it would. Now if LVGL took it a step further by calculating the actual update areas size and the bounding box size and if the bounding box size is more than 2 times the size of the updated areas then it would split the rendering into multiple smaller areas to eliminate the overhead of rendering areas that don't have anything that has updated.

Here is some example code to show what I am talking about with the bounding box.

```c
static inline void calc_box(lv_area_t *area1, lv_area_t *area2, lv_area_t *ret_area)
{
    ret_area->x1 = LV_MAX(area1->x1, area2->x1);
    ret_area->y1 = LV_MAX(area1->y1, area2->y1);
    ret_area->x2 = LV_MAX(area1->x2, area2->x2);
    ret_area->y2 = LV_MAX(area1->y2, area2->y2);
}

bool calculate_bounding_box(lv_area_t areas[], uint16_t count, lv_area_t *ret_area)
{
    if (count == 0) {
        return false;
    }

    if (count == 1) {
        memcpy(bb_area, *areas[0], sizeof(lv_area_t));
        return true;
    }

    calc_box(&areas[0], &areas[1], ret_area);

    for (uint16_t i=2;i<count;i++) {
        calc_box(ret_area, &areas[i], ret_area);
    }

    return true
}

int calculate_update_area_factor(lv_area_t areas[], uint16_t count)
{

    lv_area_t ret_area;

    if (!calculate_bounding_box(areas, count, &ret_area)) {
        return 0
    }

    uint32_t bb_area = lv_area_get_size(&ret_area);
    uint32_t area = 0;

    for (uint16_t i=0;i<count;i++) {
        area += lv_area_get_size(&areas[i]);
    }

    return (bb_area * 100) / (area * 100);
}

if (calculate_update_area_factor(areas, count) > 2) {
    // code to combine areas together in groups that have less than a 2 factor
} else {
    // everything is good how it is. calculate the bounding box and render 
    // everything inside of that bounding box 
}  
```

kdschlosser, 2025-03-18:
In those example videos you can see exactly what LVGL is going because with the smaller buffer you can see the tearing. LVGL is updating the display like this...

[image]

kdschlosser, 2025-03-18:
If I pass a partial buffer that is 1/2 the size of the frame LVGL doesn't render only this area.

[image]

It renders this area

[image]

liamHowatt, 2025-03-18:
Please enable `LV_USE_REFR_DEBUG` to see clearly which areas are being drawn.

kdschlosser, 2025-03-19:
That is not showing what is actually being drawn into the frame buffer.. It is only going to outline the objects that have been marked as invalid but there are objects that are not invalid that end up being rendered.

LVGL ONLY renders in lines that are the width of the display.

If you step through the code starting from here..

https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/core/lv_refr.c#L678

note that `LV_STRIDE_AUTO` is defined as `#define LV_STRIDE_AUTO 0`

you will see what is happening.

https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/core/lv_refr.c#L646
https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/draw/lv_draw_buf.c#L373
https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/draw/lv_draw_buf.c#L381
https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/draw/lv_draw_buf.c#L383
https://github.com/lvgl/lvgl/blob/9c043167685fc08fbcd30ddf2c285ea1089be82d/src/draw/lv_draw_buf.c#L698

so as you can see the size that is used for the rendered area is ALWAYS going to be the width of the display. It doesn't trim down the area so that ONLY the actual area that is being updated is what needs to be rendered. That is an incredibly inefficient thing to do. I mean think about it if you have a 20 pixel x 20 pixel area that needs to be updated LVGL is instead  going to render an area the size of the frame buffer to update that little tiny area. And IF the frame buffer is only large enough to hold 10 lines x the display width then it is going to need to render twice the size of the frame buffer because it is going to have to split the update. Where is LVGL simply updated ONLY that single area and not the entire width of the display not only would that update fit into a single partial buffer it would also not have to render large areas that didn't get updated. This would be a significant boost in performance. A user could also size a partial buffer to be the size of the largest area on the display that gets updated and LVGL would only use ma portion of the buffer to perform an update if the update was not an area that would be large enough to fill the buffer. On top of that you also have a much smaller amount of data being transferred to the display so you would get a much higher frame rate.

kdschlosser, 2025-03-19:
I can easily see at least a 50% increase in performance by only rendering things that have actually been updated.

kdschlosser, 2025-03-19:
I believe there is a bug in the code....

```
for(i = 0; i < tile_cnt; i++) {
            lv_area_t tile_area;
            lv_area_set(&tile_area, area_p->x1, area_p->y1 + i * tile_h,
                        area_p->x2, area_p->y1 + (i + 1) * tile_h - 1);

            if(i == tile_cnt - 1) {
                tile_area.y2 = area_p->y2;
            }

            lv_layer_t * tile_layer = &tile_layers[i];
            lv_draw_layer_init(tile_layer, NULL, layer->color_format, &tile_area);
            tile_layer->buf_area = layer->buf_area; /*the buffer is still large*/
            tile_layer->draw_buf = layer->draw_buf;
            refr_configured_layer(tile_layer);
        }
```

I believe that the code block above should read

```c
for(i = 0; i < tile_cnt; i++) {
            lv_area_t tile_area;
            lv_area_set(&tile_area, area_p->x1, area_p->y1 + i * tile_h,
                        area_p->x2, area_p->y1 + (i + 1) * tile_h - 1);

            if(i == tile_cnt - 1) {
                tile_area.y2 = area_p->y2;
            }

            lv_area_t trimmed_buf_area = {
                .x1=0,
                .y1=0,
                .x2=tile_area.x2 - tile_area.x1,
                .y2=tile_area.y2 - tile_area.y1
           };

            lv_layer_t * tile_layer = &tile_layers[i];
            lv_draw_layer_init(tile_layer, NULL, layer->color_format, &tile_area);
            tile_layer->buf_area = &trimmed_buf_area; 
            tile_layer->draw_buf = layer->draw_buf;
            refr_configured_layer(tile_layer);
        }
```

That should only allow the actual areas that have updated to be rendered. 

This is only going to be active if the user sets the number of "tiles" which seems rather strange to have to do. This should be automatically calculated based on the amount of invalid area against areas that gets rendered but is not invalid.

liamHowatt, 2025-03-21:
You're talking about for tiled rendering? That's a very important detail to mention. Maybe @kisvegabor can comment on it, as he developed tiled rendering.

Non-tiled rendering:

https://github.com/user-attachments/assets/591ab178-71eb-46c7-b945-ce5013406f68

closed 2025-04-22 by lvgl-bot