lvgl/lvgl #7958: Move demos to other repo and include as submodule

PGNetHun, 2025-03-19:
### Introduce the problem

At the moment `demos` folder size is 60 MB and it has a lot of files (assets and generated code files mainly).
Not everybody want to compile demo applications, but use only LVGL code.

### Proposal

It would be good to move `demos` out of main LVGL repo into a separate `lvgl_demos` repo, and include it as a submodule.
So when anybody clones LVGL then those demo files are excluded in the beginning... and when somebody wants to compile any demo app then simply initialize `demos` submodule and compile.

It's just my view that demo applications should not be part of main (core) repo, because LVGL codebase itself is already large enough, even without the demo files.

fbiego, 2025-03-21:
Previously, the demos were in a separate repo https://github.com/lvgl/lv_demos
Given the extra steps needed to compile demos, especially on Arduino, it seems like keeping them separate would simplify the process.

Maybe @kisvegabor @liamHowatt can share the reasoning behind merging them? This would help in understanding the feasibility of keeping them separate again.

kisvegabor, 2025-03-25:
The same idea just came up internally as well. So it's a great confirmation that something needs to be done here.

However
- some vendors are relying on having [some of the current demos in LVGL](https://github.com/lvgl/lv_port_renesas_ek-ra8d1_gcc/blob/master/src/LVGL_thread_entry.c#L27-L46) so removing them would be a breaking change for their example projects
- we can't add the `demos` as a submodule to `lvgl` as most of the library mangers (Arduino, PlatformIO, ESP-IDF don't support submodules).
- we open the versioning and maintenance issues of syncing two repos.

I suggest moving only the new and not so popular demos to another repo:
- ebike
- flex_layout
- high_res
- multilang
- scroll
- smartwatch
- transform

and keep:
- benchmark
- widgets
- music
- keypad_encoder
- vector_graphic
- stress
- render

If want to do that we should do it this week, before our release (planned for end of March, but probably will be a little later).

closed 2025-04-09 by liamHowatt

sadik-smd, 2025-09-02:
As these demos have been moved from lvgl demos 
I suggest moving only the new and not so popular demos to another repo:

ebike
flex_layout
high_res
multilang
scroll
smartwatch
transform

@kisvegabor @liamHowatt How can these can be included into the lvgl demos and built using lv_port_linux:https://github.com/lvgl/lv_port_linux
 as there is no proper docymentation for the repo lv_demos:https://github.com/lvgl/lv_demos

PR #8022 (2025-04-09): arch(demos): move new demos to lv_demos

Fixes #7958 

See the demos moved here: https://github.com/lvgl/lv_demos

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kco [... cut]