lvgl/lvgl #7962: Add Conditional File Compilation

AndreCostaaa, 2025-03-20:
### Problem to solve

Currently, compiling LVGL from scratch takes too long (try it)

The reason for this is that the conditional compilation is handled inside the .c files directly and the compiler must compile every single file even if we didn't enable a specific feature

This is specially problematic when porting LVGL to a new platform or when toying with new config options as a change to the config will result in a full compilation of the library
It is also weird seeing the compilation logs mentioning files like "thorvg" or "demo smartwatch" when those features are disabled

### Success criteria

Decrease the build time of the library

### Solution outline

This depends on #7934 which imports the configuration into cmake directly, this would allow us to not add certain files to the compilation process directly in cmake

I believe we could add a new config `LV_USE_CONDITIONAL_COMPILATION` that would enable this feature

This would only work when using cmake to compile LVGL

### Rabbit holes

Since LVGL must stay build system agnostic, we can't depend on this feature, this wouldn't replace the current #if #endif wrapping the C files, it would be something extra

One specific platform i have in mind is Arduino which AFAIK, always uses file globing to handle the compilation of libraries

### Testing

Testing would be done by checking that the correct functions were included in the final static library by simply compiling with the new flag and without and checking that both static libraries are the same

### Teaching

This can be easily added to the documentation and probably enabled by default when using cmake

kisvegabor, 2025-04-07:
In the meantime we have discussed that excluding examples, demos and thorvg if the they are no enabled seems like a good trade off, as each of these can exclude a lot of files and they are only used/supported to some extent. 

Related to https://github.com/lvgl/lvgl/pull/7934

AndreCostaaa, 2025-04-22:
Not stale, should be added after 9.3 is released

closed 2025-05-22 by lvgl-bot