lvgl/lvgl #7968: Replace SimSun Font

faxe1008, 2025-03-21:
### Introduce the problem

Brought up in our downstream fork of the module:
https://github.com/zephyrproject-rtos/zephyr/issues/87363

There is no license for SimSun to be found.

### Proposal

Replace the font with one that has a known permissive license, e.g.
https://github.com/adobe-fonts/source-han-sans/tree/master

faxe1008, 2025-03-26:
Generally speaking there are loads of fonts in the repository for which licensing is unclear:
```
./scripts/built_in_font/unscii-8.ttf
./src/libs/freetype/arial.ttf
./examples/libs/freetype/Lato-Regular.ttf
./examples/libs/freetype/NotoColorEmoji-32.subset.ttf
./examples/libs/tiny_ttf/Ubuntu-Medium.ttf
./tests/src/test_files/fonts/noto/NotoSansSC-Regular.ttf
./tests/micropy_test/font_montserrat_24.ttf
./demos/smartwatch/assets/SpaceGrotesk-SemiBold.ttf
./demos/smartwatch/assets/RobotoSerif-Light.ttf
./demos/smartwatch/assets/RobotoSerif-Medium.ttf
./demos/smartwatch/assets/RobotoSerif-ExtraBold.ttf
./demos/smartwatch/assets/Inter_28pt-ExtraBold.ttf
./demos/smartwatch/assets/SpaceGrotesk-Medium.ttf
./demos/smartwatch/assets/SpaceGrotesk-Regular.ttf
./demos/smartwatch/assets/RobotoSerif-Black.ttf
./demos/smartwatch/assets/RobotoSerif-ExtraLight.ttf
./demos/smartwatch/assets/Inter_28pt-Thin.ttf
./demos/smartwatch/assets/RobotoSerif-Bold.ttf
./demos/smartwatch/assets/SpaceGrotesk-Bold.ttf
./demos/smartwatch/assets/Inter_28pt-Light.ttf
./demos/smartwatch/assets/RobotoSerif-SemiBold.ttf
./demos/smartwatch/assets/Inter_28pt-Regular.ttf
./demos/smartwatch/assets/RobotoSerif-Regular.ttf
./demos/smartwatch/assets/SpaceGrotesk-Light.ttf
./demos/smartwatch/assets/Inter_28pt-Bold.ttf
./demos/ebike/assets/Inter-SemiBold.ttf
./demos/ebike/assets/TrumpGothicPro-Bold.ttf
./demos/ebike/assets/TrumpGothicPro.ttf
./demos/multilang/assets/fonts/NotoNaskhArabic-SemiBold.ttf
./demos/multilang/assets/fonts/NotoSerifDevanagari-SemiBold.ttf
./demos/multilang/assets/fonts/Montserrat-Bold.ttf
./demos/multilang/assets/fonts/NotoSerifHebrew-SemiBold.ttf
./demos/multilang/assets/fonts/Montserrat-SemiBold.ttf
./scripts/built_in_font/SourceHanSansSC-Normal.otf
./demos/multilang/assets/fonts/NotoSansSC-Medium.otf
```

faxe1008, 2025-03-26:
I went over the fonts a did a check (I am not a lawyer tho, be aware):

Should be OK (UBUNTU FONT LICENCE Version 1.0):
```
./examples/libs/tiny_ttf/Ubuntu-Medium.ttf
```

Rests with Microsoft seems to be forbidden to redistribute (https://learn.microsoft.com/en-us/typography/fonts/font-faq):
```
./src/libs/freetype/arial.ttf
```

Payed license:
```
./demos/ebike/assets/TrumpGothicPro-Bold.ttf
./demos/ebike/assets/TrumpGothicPro.ttf
```

OK - but no liecense file intree (SIL Open Font):
```
./demos/smartwatch/assets/RobotoSerif-Light.ttf
./demos/smartwatch/assets/RobotoSerif-Medium.ttf
./demos/smartwatch/assets/RobotoSerif-Black.ttf
./demos/smartwatch/assets/RobotoSerif-ExtraLight.ttf
./demos/smartwatch/assets/RobotoSerif-ExtraBold.ttf
./demos/smartwatch/assets/RobotoSerif-Regular.ttf
./demos/smartwatch/assets/RobotoSerif-Bold.ttf
./demos/smartwatch/assets/RobotoSerif-SemiBold.ttf
./examples/libs/freetype/Lato-Regular.ttf
./examples/libs/freetype/NotoColorEmoji-32.subset.ttf
./demos/multilang/assets/fonts/NotoSansSC-Medium.otf
./demos/multilang/assets/fonts/NotoSerifHebrew-SemiBold.ttf
./demos/multilang/assets/fonts/NotoSerifDevanagari-SemiBold.ttf
./demos/multilang/assets/fonts/NotoNaskhArabic-SemiBold.ttf
./tests/src/test_files/fonts/noto/NotoSansSC-Regular.ttf
./demos/smartwatch/assets/SpaceGrotesk-Light.ttf
./demos/smartwatch/assets/SpaceGrotesk-Regular.ttf
./demos/smartwatch/assets/SpaceGrotesk-Bold.ttf
./demos/smartwatch/assets/SpaceGrotesk-Medium.ttf
./demos/smartwatch/assets/SpaceGrotesk-SemiBold.ttf
./demos/smartwatch/assets/Inter_28pt-ExtraBold.ttf
./demos/smartwatch/assets/Inter_28pt-Thin.ttf
./demos/smartwatch/assets/Inter_28pt-Light.ttf
./demos/smartwatch/assets/Inter_28pt-Regular.ttf
./demos/smartwatch/assets/Inter_28pt-Bold.ttf
./demos/ebike/assets/Inter-SemiBold.ttf
```

To me it seems like the shipment of arial and TrumpGothicPro is potentially problematic.

kisvegabor, 2025-03-31:
I'm checking it with out designer. 

In the meantime I moved the new demos to a new repo: https://github.com/lvgl/lvgl/pull/8022