lvgl/lvgl #7981: lv_chart_set_div_line_count(chart, LV_CHART_AXIS_PRIMARY_Y, 20);  does not change the amount of lines

ats3788, 2025-03-24:
### LVGL version

v9.2.2  

### Platform

  v9.2.2  lv_chart_set_div_line_count(chart, LV_CHART_AXIS_PRIMARY_Y, 20);  does not change the amount of lines
  I tried even with 100 lines no effect at all

    ```
    lv_obj_t * chart = lv_chart_create(wrapper);
    lv_obj_set_size(chart, 300, 100); // Breite: 300px, Höhe: 100px
    lv_chart_set_type(chart, LV_CHART_TYPE_BAR); // Teste auch LV_CHART_TYPE_LINE
    lv_chart_set_range(chart, LV_CHART_AXIS_PRIMARY_Y, 0, 100);
    lv_chart_set_range(chart, LV_CHART_AXIS_SECONDARY_Y, 0, 400);
    lv_chart_set_point_count(chart, 12);

    /*Grid explizit setzen*/
    lv_chart_set_div_line_count(chart, LV_CHART_AXIS_PRIMARY_Y, 20); // 20 Linien auf Y-Achse
    lv_chart_set_div_line_count(chart, LV_CHART_AXIS_PRIMARY_X, 10); // 10 Linien auf X-Achse

    /*Grid-Stil sicherstellen*/
    lv_obj_set_style_line_width(chart, 2, LV_PART_MAIN); // Erhöhe auf 2 für Sichtbarkeit
    lv_obj_set_style_line_color(chart, lv_color_make(128, 255, 255), LV_PART_MAIN); // Grau
    lv_obj_set_style_line_opa(chart, LV_OPA_COVER, LV_PART_MAIN); // Volle Deckkraft
    lv_obj_set_style_bg_opa(chart, LV_OPA_0, LV_PART_MAIN); // Hintergrund komplett transparent
    lv_obj_set_style_pad_all(chart, 10, LV_PART_MAIN); // Padding für Sichtbarkeit
```

### What happened?

always the same amount of lines

ats3788, 2025-03-25:
What is odd I do this lv_chart_set_div_line_count(chart, 10 , 10); // 20 Linien auf Y-Achse

```
void lv_chart_set_div_line_count(lv_obj_t * obj, uint8_t hdiv, uint8_t vdiv)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);

    lv_chart_t * chart  = (lv_chart_t *)obj;
    if(chart->hdiv_cnt == hdiv && chart->vdiv_cnt == vdiv) return;
    LV_LOG_USER("Passed Chartline hdiv : %d vdiv : %d \r\n");
    chart->hdiv_cnt = hdiv;
    chart->vdiv_cnt = vdiv;
```
and I get [User]  (0.000, +0)      lv_chart_set_div_line_count: Passed Chartline hdiv : 0 vdiv : 10 
so the Question is why hdiv will not be passed

kisvegabor, 2025-04-10:
Hi,

It should be used like this:
```c
lv_chart_set_div_line_count(chart, 10, 20);
```

See
```c
void lv_chart_set_div_line_count(lv_obj_t * obj, uint8_t hdiv, uint8_t vdiv);
```

closed 2025-05-07 by lvgl-bot