lvgl/lvgl #8012: Enable builds for linux with system libraries

bkmgit, 2025-03-27:
### Problem to solve

Many linux distributions prefer to use system libraries rather than bundled libraries.  To enable this to be easily done, options should be added to the CMake configuration files to use system libraries.  The bundled libraries should also indicae the original source commit or release commit they are derived from to allow for easy checking of compatibility.

### Success criteria

Easily switch default of using bundled libraries to using system libraries.

### Solution outline

Modify CMakeLists.txt files to allow
- [barcode](https://github.com/lvgl/lvgl/tree/master/src/libs/barcode)
- [expat](https://github.com/lvgl/lvgl/tree/master/src/libs/expat)
- [gif](https://github.com/lvgl/lvgl/tree/master/src/libs/gif)
- [lodepng](https://github.com/lvgl/lvgl/tree/master/src/libs/lodepng)
- [lz4](https://github.com/lvgl/lvgl/tree/master/src/libs/lodepng)
- [qrcode](https://github.com/lvgl/lvgl/tree/master/src/libs/qrcode)
- [thorvg](https://github.com/lvgl/lvgl/tree/master/src/libs/qrcode)
- [rapidjson](https://github.com/lvgl/lvgl/tree/master/src/libs/thorvg/rapidjson)
- [tjpgd](https://github.com/lvgl/lvgl/blob/master/src/libs/tjpgd)
to be incorporated from system libraries

### Rabbit holes

Some of the modifications may not be portable or difficult to catch.  One can add some additional tests to check functionality of these libraries if they are not covered in the current tests.

### Testing

LVGL demos and tests should pass. May add additional tests if needed.

### Teaching

Main change is to document the additional build options for linux.

bkmgit, 2025-04-12:
Still of interest

AndreCostaaa, 2025-05-02:
> As some of these libraries have custom modifications/patches

Yes, that's another reason why I'm leaning against this for now. Maybe when your work in #8193 is done we may find a clean way to do it

> Wouldn't it be a great idea to use PCPP in the CI to make linking with dependencies required for tests cleaner? What do you think ?

I agree. In fact, I believe the only `cmake` variables that should exist should be the ones needed to check the type of config (`lv_conf.h` vs `.config`) and how to include that config. Once we are able to read and parse a text file config, every decision after that should come from the parsed values

--- 

Semi-related, but speaking of `pcpp`, in order to not have another issue like #8170 where the person wanted to build LVGL offline and wasn't able to do it because we always try to install `pcpp` when the script is run, I'm thinking we can change the script to handle `pcpp` the following way:

1. Check if it's installed using something like:

```python
ret = os.system("pcpp --version") 
```

2. If not try to install it using python-env (the way it's currently done)
 
This way, users wanting to build LVGL offline or that want to speed up the compilation can still install `pcpp` and for everyone else it will be installed for them

I'm not opening a PR about this as I don't want to cause conflicts with your ongoing work. Feel free to add this change in your PR if you feel comfortable doing it

bkmgit, 2025-05-03:
Is it possible to get many of the custom patches upstreamed?  This would likely reduce maintenance burden.  It would also make packaging and redistribution of Linux software that uses LVGL much easier.

bkmgit, 2025-05-22:
Can get it to build in Fedora:
https://copr.fedorainfracloud.org/coprs/fed500/lvgl/
However, we really prefer not to bundle.

The following 8 libraries are bundled
- [barcode](https://github.com/lvgl/lvgl/tree/master/src/libs/barcode)
- [expat](https://github.com/lvgl/lvgl/tree/master/src/libs/expat)
- [gif](https://github.com/lvgl/lvgl/tree/master/src/libs/gif)
- [lodepng](https://github.com/lvgl/lvgl/tree/master/src/libs/lodepng)
- [lz4](https://github.com/lvgl/lvgl/tree/master/src/libs/lodepng)
- [qrcode](https://github.com/lvgl/lvgl/tree/master/src/libs/qrcode)
- [thorvg](https://github.com/lvgl/lvgl/tree/master/src/libs/thorvg)
- [rapidjson](https://github.com/lvgl/lvgl/tree/master/src/libs/thorvg/rapidjson)
- [tjpgd](https://github.com/lvgl/lvgl/blob/master/src/libs/tjpgd)

This would mean at most 8 optional variables, the default could be to use the bundled versions.
It is understandable if the changes cannot be upstreamed, though in this case it is preferable not
to have additional functionality enabled by patches if interfaces are still compatible with the system
libraries.

bgilbert, 2025-07-22:
This seems like a long-term issue/consideration.  Is there a way to stop the bot from trying to close it every month?