lvgl/lvgl #8019: VFX in LVGL

W-Mai, 2025-03-28:
### Introduce the problem

I changed the code of LVGL's SW Renderer a little bit to achieve the following effect.

I feel super fun.

I'd like to introduce this functionality into LVGL sometime in the future when I have some free time, a way that is easily expandable, efficient, and works well. In the meantime the feature can be turned off, if not needed, thus not affecting existing functionality and code.

In fact, it's running pretty efficiently right now. But the changes are little tricky.

https://github.com/user-attachments/assets/4b148449-c60d-48c8-bc6f-e10fefd92037
https://github.com/user-attachments/assets/e247e7ed-0712-49da-a86c-058d4dd91d1a
https://github.com/user-attachments/assets/90c0eb68-87cf-408c-8b58-6c261c093dd8
https://github.com/user-attachments/assets/63eb1b4b-8d5c-42ac-9b4e-8fb10e287c72
https://github.com/user-attachments/assets/897ee310-8f71-4a17-9d38-14a7f52ee0a3

### Proposal

I changed the `transform_*` functions to recalculate the pixel position.

lizhaoming634, 2025-03-29:
LGTM

kisvegabor, 2025-03-31:
Wao, it's really interesting and fun! A few comments:
- Can you imagine some real life use cases too?
- How does it work in short?
- Can it support perspective transformation? Can we add these as hooks/callbacks or so? 
- Is it slow?
- Does it work in partial mode too?

kisvegabor, 2025-04-07:
Sounds great! Can we also use the mesh to map and LVGL image/layer as a texture? E.g. to map a text or a slider to a sphere surface (assuming that we have the required mesh)?

kisvegabor, 2025-04-14:
@djixon Thank you for the explanation but don't we need the input data a buffer in both cases. 

In case of "distortion" we say that place  `dest(x; y) = src(x + dx; y+dy)`. As `dx` and `dy` can be anything on `src` we need the whole `src` buffer.

In case of "undistortion" we have `dest(x+dx; y+dy) = src(x; y)`. So we index `src` linearly and map it to `dest` right? However if need to update only a portion of `dest` we need the whole `src` as we don't know what does where. Is it correct?

In image rotations we use the  "distortion" way, so linearly address `dest` and map the right `src` pixels.

In general I don't mind buffering `src` for complex operations like this. 

---

For the sphere text mapping I meant a mesh like this:

[image]

or the inverse (depending on how we approach it):

[image]

And if we map a text to it (assuming that the mesh is detailed enough) it can look like being on a sphere. What do you think?

djixon, 2025-04-15:
@kisvegabor 
Sure. 

Both cases require additional "displacement" buffer. The difference is only related to the nature of refreshing display. Look, at the end, when pixels data are really sent to screen, there is usually some loop (lets say, itterating through pixel color buffer by DMA controller) which read from color buffer and send it to LCD. That process is a reading of pixels color data belonging to a line, then next line and so on (of course, line is non necessary the width of the screen). At that point (from the point of view from that loop) there is no way to use the information of displacement, because nature of the stream which is send to LCD requires color information of THAT X,Y coordinate of the screen. There is no way to say something like "ok, this pixel should be drawn shifted by (dx,dy)" because that DATA in the stream represent x,y coordinate on the screen. In data seqence of that loop you need to provide the color of THAT pixel with THAT coordinate on the screen.  

However, in second case it is possible. Also from the point of view from that loop, interpretation of that second case is like "ok, THAT x,y pixel contains color from the color buffer at (x+dx, y+dy)" so order of the pixels drawn on the screen is preserved, and only data is pulled from different buffer location. 

That is the key difference. 

That second case allows the displacement is performed at the lowest possible level, inside that loop which iterattes through color buffer (by replacing colors just by reading data from other position in buffer on-the-fly). In the first case, you do not know the color of "actual" pixel which has to be drawn, as you can not calculate it from that location, only you have (at that buffer location, with respect to that loop) is "where that actual pixel should be drawn" but stream requires to be populate with color of actual X,Y pixel.  

Conclusion: Second case represents a color function of actual screen coordinate (X,Y) which can be called from streaming loop something like CurrColor=GetColorData(X+dx,Y+dy). However, the first case can not be written as the function of (X,Y) coordinate, because color of pixel at X,Y does not depend on X,Y !!!! At the X,Y you simply do not know what other possible locations could be displaced at THIS X,Y coordinate with their coresponding colors.  So, second case is much easier for implementation as it requires only that single displacement buffer (and color buffer of course). The nature of the first case is such that it beside displacement and color buffer requires one more additional temporary buffer for computation, which requires initialization to black (transparent, if we are talking about layers) and then populated by weighted color sums from color buffer using diplacement buffer.....

Sure. Text or any image, can look like deformed in any way, in general, 2D displacement buffer can be reduced to half or quorter of color buffer resolution (in some cases even more) and interpolation performed on-the-fly during processing. 2D displacement buffer, lets call it as it is properly called in "proffesional video editing world", UV remapping buffer, does not contain "graphic". It contains dx,dy information. So data can be negative too and that must be taken into account for making decision of data format which will hold that information. For example, if only two bytes (16 bits), one byte for dx and other for dy, are used, remapping is possible in range of +-128 pixels only, with resolution of 1 pixel (which is not enough for serious usage).  By doubling it, would be 4 bytes per pixel which might be overhead for some cases. Using second case, brings one more benefit. Since second case IS A COLOR FUNCTION OF X,Y it allows using absolute coordinates. For example, starting rectangular color buffer, can be normalized (in coordinate sense) and U,V computed on-the-fly as U=X/(width-1) and V=Y/(height-1), and UV remapping buffer contains the information in normalized floating values. That way you have universal defformation buffer which can be used for transformation of any resolution rectangular color buffer (of course, resolution aspect can change) because any rectangle can be normalized in (0.0f-1.0f) coordinate space...... But anyway, with many years of experience behind me, I can say, there are other more important things which should be implemented in LVGL with much greater level of priority. After them, UV remapping "comes" naturally. We can talk about them through email or live. I will visit the Budapest during the following Easter and if you are availble we can organize some casual meeting and talk about possible improvements on the library.

kisvegabor, 2025-04-15:
> I will visit the Budapest during the following Easter and if you are availble we can organize some casual meeting and talk about possible improvements on the library.

It's amazing! I'd be very happy to meet you! Please write me an email to gabor@lvgl.io to discuss the details. 

---

So practical example:
We have an `src` buffer like this
```
1 2 3 
4 5 6 
```

We want to transform it a little so we have this displacement map (just to shift the first line):
```
-1,0  -1,0  -1,0  -1,0
 0,0   0,0   0,0   0,0
```

Following the formula of `dest[x, y] = src[x + map[x,y].x, y + map[x,y].y]` we get
```
0 1 2 3
4 5 6 0
```

Obviously we need to fractional numbers to handle antialiasing.

djixon, 2025-04-15:
In general, yes, that is the principle. However, there is a way to avoid negative values if "2D displacement" is determined in absolute space (in that case it is called "UV mapping"). UV coordinate map contains the exact coordinate from the normalized src buffer. Any rectangular area can be thought with floating point pixel coordinates ranges 0-1 for whole width of buffer and 0-1 for whole height of buffer. Doesnt matter of the buffer resolution, each pixel can be represented in such space by computing float x = X / (Width-1) and float y= Y / (Height-1). Deformation buffer contains values, from that space. So by drawing deformed image on the screen (or another layer, or other buffer) iteratting is performed over deformation buffer (UV map) and each pixel of UV map contains pair of cordinates from the normalized src color buffer. In such case

dest[X, Y] = src[UVmap[X, Y].x * (Width-1) , UVmap[X, Y].y * (Height-1)]     

There are other more important things which should be applied first. Some of them are support for floats or half-floats by keeping them as integers whole the time. But that is complete another topic. To get a "picture" of why that is important take for an example this task. The goal is to make single, lets say red pixel in smooth motion in, lets say 30 frames, from coordinate 10,10 to coordinate 11,10 (horizontal shift of 1 pixel in 30 frames). In integer coordinate space it looks impossible, but that is not true. Suppose that initial zero frame contains full red pixel at 10,10 and full black pixel at 11,10. at frame 29 pixel at 11,10 will be red while pixel ax 10,10 will be black. But what about, ie. frame 15. Well, at frame 15 our pixel should be at imaginarry coordinate 10.5,10 but that coordinate doesnt exists in integer coordinate space. Now, look at the fractional part of that pixel. It is 0.5 abd it is at that value because it should cover only 50% of the starting pixel location and 50% of the destination pixel location. So we can take that value 0.5, multiply color of our pixel (in this case Red) and store it to both locations 10,10  and 11,10 in frame 15.  Similary, at frame 20 we "moved" 2/3 from the first pixel and covering 1/3 of destination pixel and so on. So without changing single line of library code we can implement for example a function: TColor  GetPixelColor(float x, float y). It would work on that principle. In the worse case 4 pixels will be involved. Those 4 are with following coordinates: [trunc(x), trunc(y)]  ,  [trunc(x)+1, trunc(y)] , [trunc(x), trunc(y)+1]  ,  [trunc(x)+1, trunc(y)+1] .  Those are 4 pixels with integer coordinates. Now we have two fractional parts which will be used as proper weights for mixing colors of those 4 pixels. Non optimized code, just for ilustration purpose would look like: (assumption is 1st quadrant for screen.... meaning 0,0 coordinate of the screen is at bottom left corner, this of course doesn't matter, but names chosen for variables reflect such screen orientation)

fx = fract(x)
fy = fract(y)
x = trunc(x)
y = trunc(y)
temp_bottom_interpolated = color[x, y]     * (1.0 - fx) + color[x+1, y]     * fx;
temp_top_interpolated       = color[x, y+1] * (1.0 - fx) + color[x+1, y+1] * fx;
// now we just need to interpolate those two color values by fractional y part
result = temp_bottom_interpolated * (1.0 - fy) + temp_top_interpolated * fy 

So, result contains weighted color mix of 4 pixels by using fractional parts from desired coordinate at floating point location. Similary, writting can be performed, mixing, deforming, antialiasing and anything which requires "data" at floating coordinates.....  

This is ilustrative example. It is also possible make it in pure integer math, but that is completely different topic.

djixon, 2025-04-23:
Sure. 
You, chosed a function which can be computed on-the-fly (however, it requires floats, some deformations are more costly on computations, ie. Laplacian flow, convolution, curl etc.  and are usually handled by deformation maps when real-time behavior is required, because it can be baked in maps and completed with integer math). 

Regardless of naming convention, the key point is to always make DEST(X,Y) pixel color computable as the function from SINGLE SAMPLE coordinate (if possible) of SRC. So technically, accessing to SRC is via reverse function. In math, if  DST = SRC * T, where T is any kind of transformation, then DST * T ^ (-1) = SRC. Thats in theory and it is related to real numbers, because any range of real numbers, doesn't matter how small or large is isomporphic to range 0.0-1.0 or any other range. With integers (pixel coordinates are integers) things change, because in some types of transformations, calc ends with many SRC pixels converge to lets say single integer pixel in DEST. For accurate results you would need to interpolate all of them into that destination pixel color. Without interpolation, in one frame one pixel (of many) can be  rendered in DST, while in next frame some other pixel color, which will produce flickering in all those areas where convergence is presented. 

Another problem, of general case, (which is not visible in your example as your example represents a smooth transformation) is in areas where T transformation separates neigbour pixels in more than single pixel (lets say 10 or even more pixels). In those areas, such trnsformation form a holes (not colored, if appropriate interpolation is not applied). Simple example of that case would be lets say, rotation of the center of image where the speed of rotation drops with radius, and pixels move toward center of rotation and border pixels of SRC image are zero transformed (fixed in place). After N rotations, inner pixels will start separating from those at border (transformation is defined that way) producing holes ......   

To avoid that and also to avoid interpolation of many pixels, once the DST is expressed as the function of (fake) floats, you are actually mimicing isomorphism which is presented among reals, by accessing single (fake) pixel color at (fake) real coordinate and grab that interpolated color which will end in DEST buffer, solving those issues of interpolations or "holes" by single smart approach.

W-Mai, 2025-04-23:
> You, chosed a function which can be computed on-the-fly (however, it requires floats, some deformations are more costly on computations, ie. Laplacian flow, convolution, curl etc. and are usually [...]

In this case, we can generate a replacement map in advance, and then access the replacement map via callback.

Now lvgl is linear interpolation. If a better effect is desired, bilinear interpolation needs to be supported.

Meanwhile, I think if you want to calculate how large the transformed area is, you don't need to worry. The simplest way to handle it is to let the user decide how large the clip area is.

I think this is a simple graphics library of ours. There is no need to pursue perfect mathematical solutions too much. A little flaw is acceptable.

It's cool enough to have this effect on a mini device I think. 😁

djixon, 2025-04-23:
Sure, for those smooth transformations which can be computed on-the-fly, you can add more simple functions like: whole line  horizontal displacement by sine (and its replicas for vertical displacement of whole column), can be interesting once a user release component to "simulate" like it bounces of floor with left/right parts going up and down several times until they settle in the middle etc.  Also, some spherical (like magnifier glass) deformation can be added as global option, where it would be applied at area where user approached touch pannel by his finger but didnt actually touched it (it is possible only with capacitive touch panel, which can sence presence of finger while it didn't touch the surface) to activate magnification geformation around finger while it is above the screen but not touching the screen etc.....

kisvegabor, 2025-04-24:
>  to activate magnification deformation around finger while it is above the screen but not touching the screen etc.....

Woooow, what a fantastic idea! We can apply something similar on a button for example. When it pressed it will be blended like it where gum, and reverts in a second. I love it! 

If all these are callbacks on images (LVGL automatically creates images from widgets on specific effects) I think it's flexible enough and the users can decide where to draw the line between pregenerated maps and runtime math. 

The callback could be treated as a style property in the end.

djixon, 2025-04-24:
Unfortunatelly, many official touch drivers ignore all events on touch panel until touch panel is actually touched. I tested it with SC01 PLUS (model ZX3D50CE02S-V13-USRC board with forced read of touch panel data and it changes X and Y bits when finger is 3-5 mm above the panel while bits for "touch down" are not active. (that board has an old touch panel which can recognize only two fingers). I guess many of capacitive touch panels do the same, but if those events do not trigger change it means its limitation is inside the driver. In that case because driver filters only events while "touch down"  exists or it is changed, simple manual call for reading of touch panel data reveals that X and Y follow the finger even if it is a few millimeters above the panel.
    
Idea about magnification can be advanced even further. It can act on specific components (for example, buttons, sliders etc) as an additional style element, or as a configurable global on-the-top layer (like overlay on the whole screen) to magnify anything (final screen for example), around finger in some radius, where all pixels of viewing screen arround the finger, regardless of which component they belong, text, image, button or anything else are simply magnified in circular area (using on-the-fly computation or baked constant displacement map prepared to generate desired deformation), allowing user to see in details anything shown on the screen...

closed 2025-05-16 by lvgl-bot