lvgl/lvgl #8021: Bug in drawing multiple lines with“ lv_scale_set_line_needle_value”

mgdsun2000, 2025-03-31:
### LVGL version

V9.1

### Platform

stm32F103

### What happened?

The **_lv_scale_set_line_needle_value_** function cannot be used when there are multiple lines in one interface (such as an analog clock). Otherwise, only the last line will be displayed, and the others will be overwritten. The reason is that the function uses static lv_point_precise_t needle_line_points[2];. This should be modified to accept an external parameter or use dynamic memory allocation.

### How to reproduce?

 lv_scale_set_major_tick_every(ui->main_ui_analog_clock_1, 5);
    lv_scale_set_text_src(ui->main_ui_analog_clock_1, main_ui_analog_clock_1_hour_ticks);
    lv_obj_update_layout(ui->main_ui_analog_clock_1);
    ui->main_ui_analog_clock_1_hour_needle = lv_line_create(ui->main_ui_analog_clock_1);
    lv_obj_set_style_line_width(ui->main_ui_analog_clock_1_hour_needle, 4, LV_PART_MAIN);
    lv_obj_set_style_line_color(ui->main_ui_analog_clock_1_hour_needle, lv_color_hex(0x00ff00), LV_PART_MAIN);
    lv_obj_set_style_line_rounded(ui->main_ui_analog_clock_1_hour_needle, true, LV_PART_MAIN);   
   lv_scale_set_line_needle_value(ui->main_ui_analog_clock_1, ui->main_ui_analog_clock_1_hour_needle, 30, main_ui_analog_clock_1_hour_value * 5);		
    ui->main_ui_analog_clock_1_min_needle = lv_line_create(ui->main_ui_analog_clock_1);
    lv_obj_set_style_line_width(ui->main_ui_analog_clock_1_min_needle, 3, LV_PART_MAIN);
    lv_obj_set_style_line_color(ui->main_ui_analog_clock_1_min_needle, lv_color_hex(0x4b00ff), LV_PART_MAIN);
    lv_obj_set_style_line_rounded(ui->main_ui_analog_clock_1_min_needle, true, LV_PART_MAIN);		
    lv_scale_set_line_needle_value(ui->main_ui_analog_clock_1, ui->main_ui_analog_clock_1_min_needle, 40, main_ui_analog_clock_1_min_value,);		
    ui->main_ui_analog_clock_1_sec_needle = lv_line_create(ui->main_ui_analog_clock_1);
    lv_obj_set_style_line_width(ui->main_ui_analog_clock_1_sec_needle, 2, LV_PART_MAIN);
    lv_obj_set_style_line_color(ui->main_ui_analog_clock_1_sec_needle, lv_color_hex(0xE1FF00), LV_PART_MAIN);
    lv_obj_set_style_line_rounded(ui->main_ui_analog_clock_1_sec_needle, true, LV_PART_MAIN);
    lv_scale_set_line_needle_value(ui->main_ui_analog_clock_1, ui->main_ui_analog_clock_1_sec_needle, 60, main_ui_analog_clock_1_sec_value,);

C47D, 2025-04-04:
Can you send an snippet with minimal code to reproduce the issue and an screenshot of the issue? I don't quite get the problem.

mgdsun2000, 2025-05-01:
Sorry, I've been really busy recently and didn't check my emails in time. When using the lv_scale_set_line_needle_value function to draw multiple needles on one UI, only one needle will be displayed. According to the code provided above(Link: https://github.com/lvgl/lvgl/issues/8021#issue-2959304090), when drawing an analog clock, only the last - drawn clock hand will always be shown.
     The root cause of the problem is that the value of static lv_point_precise_t needle_line_points[2] in the lv_scale_set_line_needle_value function will be overwritten by subsequent operations. As a result, the multiple needles being drawn end up being just the last one.

EduardoGMohedano, 2025-05-22:
Hi @mgdsun2000 could you add a snippet of code to further explain your issue. I just ran analog clock example in lvgl 9.3 version and it seems it works properly as it uses two different needles for hour and minutes. Attached is the image for reference. Maybe if you share your complete code I can deep dive into the issue if I can replicate it first

[image]