lvgl/lvgl #8037: spangroup doesn't grow height when height set to LV_SIZE_CONTENT

ino-josh, 2025-04-01:
### LVGL version

9c043167685fc08fbcd30ddf2c285ea1089be82d

### Platform

SDL on Linux

### What happened?

When a spangroup's height is set to LV_CONTENT_SIZE, it doesn't actually grow its height to fit its contents.

### How to reproduce?

```c
  lv_obj_t* spans = lv_spangroup_create(parent);
  lv_obj_set_size(spans, LV_SIZE_CONTENT, LV_SIZE_CONTENT);

  for (size_t i = 0; i < 4; i++) {
    lv_span_t* span = lv_spangroup_add_span(spans);
    lv_span_set_text_static(span, "This is a test\n");
  }
  lv_spangroup_refresh(spans);
```

You would expect to see this:
```
This is a test
This is a test
This is a test
This is a test
```

but it looks like this:
[image]

ino-josh, 2025-04-29:
Not abandoned, not stale. Still an issue.

AndreCostaaa, 2025-05-02:
Hi @ino-josh 

The reason is because the spangroup size is not updated if BOTH the width and the height are set to `LV_SIZE_CONTENT`.
Since you want a fixed width and the text to wrap, you should set a fixed width

```c
lv_obj_t *spans = lv_spangroup_create(lv_screen_active());
lv_obj_set_size(spans, lv_pct(100), LV_SIZE_CONTENT);

for (size_t i = 0; i < 4; i++) {
	lv_span_t *span = lv_spangroup_add_span(spans);
	lv_span_set_text_static(
		span, "This is a very huge huge huge huge huge text\n");
}
lv_spangroup_refresh(spans);
```

ino-josh, 2025-05-02:
Why shouldn't it update the size?
In my example, each line has a newline at the end. So it isn't wrapping (not a fixed width), it is placed on a new line.
So the width should automatically be as wide as the widest line, and the height should be enough to show all the lines.
That is the expected result, in my opinion. It is what would happen if the text were in a `label` instead of a `spangroup`.

closed 2025-05-27 by lvgl-bot