lvgl/lvgl #8039: Change lv_draw_sw_init thread init with variable thread priority

MootSeeker, 2025-04-02:
### Introduce the problem

In the function `void lv_draw_sw_init(void)`, a thread is created as follows:

```
#if LV_USE_OS
        lv_thread_init(&draw_sw_unit->thread, LV_THREAD_PRIO_HIGH, render_thread_cb, LV_DRAW_THREAD_STACK_SIZE, draw_sw_unit);
#endif
```

This occurs because I configured the following in lv_conf.h:
```
#define LV_USE_OS       LV_OS_FREERTOS
#define LV_USE_DRAW_SW  1
```
However, my embedded system is designed with a completely event-driven, non-blocking architecture and thus requires no more than three priority levels. As a result, the lv_init() function always crashes because it attempts to create a task using an unavailable priority level.

### Proposal

A more robust solution would be to make this thread priority configurable directly within the configuration file, especially since I explicitly define that tasks will be used.

Here is an example of how this could be implemented in the configuration:

```
#define LV_USE_DRAW_SW 1
#if LV_USE_DRAW_SW == 1

/* Add configurable thread priority here */
#define LV_DRAW_SW_THREAD_PRIORITY LV_THREAD_PRIO_HIGH

/* Existing configuration options */
#define LV_DRAW_SW_SUPPORT_RGB565       1
#define LV_DRAW_SW_SUPPORT_RGB565A8     0
#define LV_DRAW_SW_SUPPORT_RGB888       0
#define LV_DRAW_SW_SUPPORT_XRGB8888     0
#define LV_DRAW_SW_SUPPORT_ARGB8888     0
#define LV_DRAW_SW_SUPPORT_L8           0
#define LV_DRAW_SW_SUPPORT_AL88         0
#define LV_DRAW_SW_SUPPORT_A8           0
#define LV_DRAW_SW_SUPPORT_I1           0

#define LV_DRAW_SW_I1_LUM_THRESHOLD 127
#define LV_DRAW_SW_DRAW_UNIT_CNT    1
#define LV_USE_DRAW_ARM2D_SYNC      0
#define LV_USE_NATIVE_HELIUM_ASM    0
#define LV_DRAW_SW_COMPLEX          0

#if LV_DRAW_SW_COMPLEX == 1
#define LV_DRAW_SW_SHADOW_CACHE_SIZE 0
#define LV_DRAW_SW_CIRCLE_CACHE_SIZE 4
#endif

#define LV_USE_DRAW_SW_ASM     LV_DRAW_SW_ASM_NONE

#if LV_USE_DRAW_SW_ASM == LV_DRAW_SW_ASM_CUSTOM
#define LV_DRAW_SW_ASM_CUSTOM_INCLUDE ""
#endif

#endif
```

MootSeeker, 2025-04-17:
@kisvegabor,

while integrating LVGL 9 on an event‑driven FreeRTOS system that exposes only three priority levels, I hit a crash in ```
lv_init():
lv_draw_sw_init()
``` always creates the draw‑SW thread with `LV_THREAD_PRIO_HIGH`, which is out of range for my target. Because the priority is hard‑coded, I can’t remap it without patching LVGL.

Would you be open to making the draw‑SW thread priority user‑configurable in lv_conf.h? Something like I mentioned before? 
This keeps the current default but lets projects with restricted RTOS priority schemes pick a suitable value.

kisvegabor, 2025-04-25:
Hi,

Sorry for the delay. I see the problem and I agree with the proposed solution, just please make the config option generic similar to `LV_DRAW_THREAD_STACK_SIZE`. E.g. `LV_DRAW_THREAD_PRIO` and update all draw units. 

Please also an option to configure it in Kconfig.

closed 2025-05-15 by kisvegabor