lvgl/lvgl #8044: Inconsistent thread sanitizer test runs with failures in lvgl code paths

phelter, 2025-04-03:
### LVGL version

v9.2.2

### Platform

VSCode under Ubuntu 24.04 Linux. Running unit tests with googletest that incorporate lvgl using X11.

### What happened?

Have a series of thread sanitizer tests that exercise LVGL code to ensure that LVGL is thread safe.

All actions are perfromed in the LVGL thread but under some circumstances there is the following error:

```
608: ThreadSanitizer:DEADLYSIGNAL
608: ==5619==ERROR: ThreadSanitizer: SEGV on unknown address 0x7fffe7a21000 (pc 0x7ffff73f3599 bp 0x7fffebbfe8e0 sp 0x7fffebbfe828 T5622)
608: ==5619==The signal is caused by a READ memory access.
608:     #0 __memcmp_avx2_movbe string/../sysdeps/x86_64/multiarch/memcmp-avx2-movbe.S:415 (libc.so.6+0x188599) (BuildId: 42c84c92e6f98126b3e2230ebfdead22c235b667)
608:     #1 MemcmpInterceptorCommon(void*, int (*)(void const*, void const*, unsigned long), void const*, void const*, unsigned long) <null> (UnitTest_ui_osd+0x922ff) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #2 memcmp <null> (UnitTest_ui_osd+0x924e4) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #3 lv_memcmp /...build/default/tsan/_deps/lvgl-src/src/stdlib/clib/lv_string_clib.c:55:12 (UnitTest_ui_osd+0x1e85d9) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #4 decoder_info /...build/default/tsan/_deps/lvgl-src/src/libs/libpng/lv_libpng.c:117:16 (UnitTest_ui_osd+0x1d81b1) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #5 image_decoder_get_info /...build/default/tsan/_deps/lvgl-src/src/draw/lv_image_decoder.c:343:31 (UnitTest_ui_osd+0x1c83e4) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #6 lv_image_decoder_get_info /...build/default/tsan/_deps/lvgl-src/src/draw/lv_image_decoder.c:86:36 (UnitTest_ui_osd+0x1c81c1) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #7 lv_draw_image /...build/default/tsan/_deps/lvgl-src/src/draw/lv_draw_image.c:107:23 (UnitTest_ui_osd+0x1c5322) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #8 draw_image /...build/default/tsan/_deps/lvgl-src/src/widgets/image/lv_image.c:794:13 (UnitTest_ui_osd+0x20157f) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #9 lv_image_event /...build/default/tsan/_deps/lvgl-src/src/widgets/image/lv_image.c:684:9 (UnitTest_ui_osd+0x20157f)
608:     #10 lv_obj_event_base /...build/default/tsan/_deps/lvgl-src/src/core/lv_obj_event.c:89:5 (UnitTest_ui_osd+0x1ab254) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #11 event_send_core /...build/default/tsan/_deps/lvgl-src/src/core/lv_obj_event.c:364:11 (UnitTest_ui_osd+0x1ab254)
608:     #12 lv_obj_send_event /...build/default/tsan/_deps/lvgl-src/src/core/lv_obj_event.c:67:23 (UnitTest_ui_osd+0x1ab254)
608:     #13 lv_obj_redraw /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:116:5 (UnitTest_ui_osd+0x1bcb6e) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #14 refr_obj /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:1032:9 (UnitTest_ui_osd+0x1bd1d7) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #15 lv_obj_redraw /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:167:21 (UnitTest_ui_osd+0x1bccac) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #16 refr_obj /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:1032:9 (UnitTest_ui_osd+0x1bd1d7) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #17 refr_obj_and_children /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:820:5 (UnitTest_ui_osd+0x1be15f) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #18 refr_area_part /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c (UnitTest_ui_osd+0x1bdd2e) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #19 refr_area /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:679:9 (UnitTest_ui_osd+0x1bc3d7) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #20 refr_invalid_areas /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:586:13 (UnitTest_ui_osd+0x1bc3d7)
608:     #21 lv_display_refr_timer /...build/default/tsan/_deps/lvgl-src/src/core/lv_refr.c:398:5 (UnitTest_ui_osd+0x1bc3d7)
608:     #22 lv_timer_exec /...build/default/tsan/_deps/lvgl-src/src/misc/lv_timer.c:326:59 (UnitTest_ui_osd+0x1e71ad) (BuildId: 382b8132a0413d471a3d497a3d940e0a0d4d786b)
608:     #23 lv_timer_handler /...build/default/tsan/_deps/lvgl-src/src/misc/lv_timer.c:107:16 (UnitTest_ui_osd+0x1e71ad)
```

So this is accessing `image_decoder_get_info` and trying to compare data against an invalid read access location.

### How to reproduce?

The setup and creation of this is quite extensive so don't have a specific test case. Checking this data path within the lvgl code base for potential read access to pointers that may be invalid is suggested.  Invalid pointers may happen due to storage of objects as references but not updating those references once they are deleted.

uLipe, 2025-04-17:
@phelter could you point what is the unit test that falls into this error?

phelter, 2025-04-19:
There is no unit test that shows this at the moment this is a unit test for an application being developed for linux that consumes lvgl and instead of mocking lvgl's extensive API - it uses it directly (no mock).

I will try and reproduce a test case for this.  I thought you might want to check the code and create unit tests for this area of the code - eg: lv_image_decoder_get_info and below and identify if there are any cases where it is used after free.

closed 2025-05-13 by lvgl-bot