lvgl/lvgl #8046: cmake throws error on Ubuntu build

pfvogel, 2025-04-03:
### LVGL version

v9.3.0

### Platform

Ubuntu 22.04

### What happened?

Initially, the cmake failed to find "pcpp" (which was not required previously).  I installed it using "apt install pcc" and now cmake fails with

```
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- The CXX compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
error: unknown option '--passthru-defines'
Error during preprocessing: Command '['pcpp', '-o', '/home/paul/lvgl/build/tmp.h', '--passthru-defines', '--line-directive=', '/home/paul/lvgl/src/lv_conf_internal.h', '-I/home/paul/lvgl', '-I/home/paul/lvgl/..', '-I/home/paul/lvgl/src']' returned non-zero exit status 8.
CMake Error at env_support/cmake/custom.cmake:72 (message):
  preprocess_lv_conf_internal.py failed with return code: 1
Call Stack (most recent call first):
  CMakeLists.txt:23 (include)

```

Reverting this commit
```
commit 371c21bbdaa8216f2a0bef0069db7ae1a727d80d
Author: David Truan <david.truan@edgemtech.ch>
Date:   Thu Apr 3 14:42:18 2025 +0200

    arch(cmake): add native Kconfig support to cmake (#7934)
```

lets the current head build without the error

### How to reproduce?

```
mkdir build ; cd build ; cmake ..
```

grybouilli, 2025-04-04:
Encountered the exact same issue just now

AndreCostaaa, 2025-04-04:
Hi there !

The `pcpp` dependency can be installed using `pip`:

```bash
pip install --user pcpp
```
Which can be seen in the diff of the commit provided as there was the consequent change to [scripts/install-prerequisites.sh](https://github.com/lvgl/lvgl/blob/master/scripts/install-prerequisites.sh)

---

You also need to make sure to update your `lv_conf.h` as some old configs had to be rewritten. For this, you have a couple of options:

If you're using a repository like `lv_port_pc_vscode` you can simply `git pull` the latest changes there as the `lv_conf.h` file has been updated for you already.

If you're using a `lv_conf.defaults` file you can regenerate the new `lv_conf.h` using the script inside `lvgl/scripts/generate_lv_conf.py`

```bash
python scripts/generate_lv_conf.py --defaults <path_to_lv_conf.defaults> --config <path_to_new_lv_conf.h
```

You can also do it manually if none of the options work for you, here's the patch:

```patch
diff --git a/lv_conf.h b/lv_conf.h
index 8734f2d..d62a812 100644
--- a/lv_conf.h
+++ b/lv_conf.h
@@ -1246,7 +1246,11 @@
 #define LV_USE_ST7796        0
 #define LV_USE_ILI9341       0
 
-#define LV_USE_GENERIC_MIPI (LV_USE_ST7735 | LV_USE_ST7789 | LV_USE_ST7796 | LV_USE_ILI9341)
+#if (LV_USE_ST7735 | LV_USE_ST7789 | LV_USE_ST7796 | LV_USE_ILI9341)
+    #define LV_USE_GENERIC_MIPI 1
+#else
+    #define LV_USE_GENERIC_MIPI 0
+#endif
 
 /** Driver for Renesas GLCD */
 #define LV_USE_RENESAS_GLCDC    0
```

pfvogel, 2025-04-04:
I had installed pcpp through "apt install pcc" so I removed this and then did the pip install instead.  
```
$ pcpp --version
pcpp 1.30
```
I usually generate the lv_conf.h by copying the lv_conf_template.h and then editing to change the "#if 0" at the top to "#if 1", so I already had the conf file changes.

The cmake step still fails, but I get a slightly different error:

```
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
../src/lv_conf_internal.h:60 error: Include file '../../lv_conf.h' not found
Error during preprocessing: Command '['pcpp', '-o', '/home/paul/lvgl/build/tmp.h', '--passthru-defines', '--line-directive=', '/home/paul/lvgl/src/lv_conf_internal.h', '-I/home/paul/lvgl', '-I/home/paul/lvgl/..', '-I/home/paul/lvgl/src']' returned non-zero exit status 1.
CMake Error at env_support/cmake/custom.cmake:72 (message):
  preprocess_lv_conf_internal.py failed with return code: 1
Call Stack (most recent call first):
  CMakeLists.txt:23 (include)
```

The lv_conf.h file is at the same location as always, so not sure why it is not being found.  And as before, if I do

```
git revert 371c21bbdaa8216f
```

to revert that one commit, the cmake and build works without errors.

Thanks for the suggestions!

AndreCostaaa, 2025-04-06:
I was able to build it successfully using your setup and the fix in #8052 

```bash
pwd ; mkdir build ; cd build ; cmake ..
/home/andre/dev/lvgl
-- The C compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- The CXX compiler identification is GNU 14.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /home/andre/dev/lvgl/tests/env/bin/python3.13 (found version "3.13.2") found components: Interpreter 
Preprocessing completed. Output saved to /home/andre/dev/lvgl/build/tmp.h
Indentation removed. Cleaned output saved to /home/andre/dev/lvgl/build/lv_conf_expanded.h
Temporary preprocessed file /home/andre/dev/lvgl/build/tmp.h removed.
-- Configuring done (1.4s)
-- Generating done (0.2s)
-- Build files have been written to: /home/andre/dev/lvgl/build
```

Could you try it and confirm it also works on your end ? :)

pfvogel, 2025-04-07:
Yep -- cmake / build working now -- thanks!!

PR #8052 (2025-04-09): build(custom): fix include path of lv_conf.cmake
Fixes #8046 

The reason the CI wasn't failing is because the `CMAKE_CURRENT_BINARY_DIR` and `CMAKE_BINARY_DIR` point to the same path when building lvgl alone : 

```bash
#inside lvgl
mkdir build; cmake ..
```

But in other repositories like the `lv_port_pc_vscode`, these paths differ as `CMAKE_CURRENT_BINARY_DIR` will be `lv_port_pc_vscode/build/lvgl` whereas `CMAKE_BINARY_DIR` will be `lv_port_pc_vscode/build` when doing something like

```bash
#inside lv_port_pc_vscode
mkdir build; cmake ..
```

cc @Daxidz
closed 2025-04-09 by kisvegabor