lvgl/lvgl #8056: Monochrome display support on LVGL V9.2

mmurty, 2025-04-05:
### Introduce the problem

Hello, 
I am trying to implement the flush_cb() function for a monochrome display from Newhaven: NHD-C128128BZ-FSW-GBW which makes use of the ST7528 display controller. 
All the online articles that I am able to grab hold of provide information for LVGL V8.x. I am trying to set the display up with LVGL V9.2. As a lot of things have changed between LVGL V8 & V9 I am not able to follow the same method for implementing the flush callback. 
The latest documentation mentions that LVGL V9 renders frame buffers for monochrome displays. I am just not able to get hold of how to interpret the framebuffer to implement the flush_cb(). 
Can anyone provide a brief idea?

SoulPrudya, 2025-04-05:
Hello,

The monochrome format assumes that every byte of px_map stores 8 pixels (1 pixel per bit). 

In my case, I have implemented a function that writes one pixel to my buffer. Therefore, in my_flush_cb I need to prepare that one pixel. 

LVGL reserves 2 x 4 bytes in the buffer for a pallet. To skip it, use `px_map += 8`.

```c
void my_flush_cb(lv_display_t * display, const lv_area_t * area, uint8_t * px_map)
{
	SSD1306_Fill(0);

	px_map += 8;

	uint32_t width = (area->x2 - area->x1 + 1);
	for (int y = area->y1; y <= area->y2; y++)
	{
		uint32_t offset = (y * width);

		for (int x = area->x1; x <= area->x2; x++)
		{
			uint32_t index = offset + x;
			uint32_t byte_index = index / 8;
			uint32_t bit_index = index % 8;

			uint8_t color = (px_map[byte_index] >> (7 - bit_index)) & 0x01;

			SSD1306_DrawPixel(x , y, color ? SSD1306_COLOR_BLACK : SSD1306_COLOR_WHITE);
		}
	}

	SSD1306_UpdateScreen();

	lv_display_flush_ready(display);
}
```

area->x1 and area->y1 don't always start from 0. so it's better to add calculations like this:

```
uint32_t offset = ((y - area->y1) * width);
uint32_t index = offset + (x - area->x1);
```

I deliberately omitted those details in the code for the sake of simplicity. Additionally, I swapped black and white for better appearance on my screen.

Your interaction with the screen and your implementation may differ, perhaps you need to use the entire byte or several bytes instead of individual bits.

SoulPrudya, 2025-04-05:
Additional information: https://docs.lvgl.io/master/details/main-modules/display/color_format.html#monochrome-displays

alexriegler12, 2025-04-12:
Maybe also add Back a monochrome mode that uses a byte per pixel (even if it is inefficient)

mmurty, 2025-04-17:
@SoulPrudya Thanks for your inputs. I got busy with the bring up of other components of the board. Soon I will come to the LVGL bring up in the next week or so and will let you know my updates.

mmurty, 2025-04-25:
@SoulPrudya Thanks for your inputs I was at least able to display hello world partially.

[image]

Its still not perfect though. 
Here is my flush_cb() implementation:
```
static uint8_t frameBuff[2048+8];
/* Buffer for one page of data */
static uint8_t pageBuff[128 * 4];

/*-------------------------
 * Initialize your display
 * -----------------------*/
ST7528Init();

/*------------------------------------
 * Create a display and set a flush_cb
 * -----------------------------------*/
lv_display_t * disp = lv_display_create(ST7528_HOR_RES, ST7528_VER_RES);
lv_display_set_flush_cb(disp, disp_flush);
lv_display_set_buffers(disp, frameBuff, NULL,
	sizeof(frameBuff), LV_DISPLAY_RENDER_MODE_FULL);

lv_theme_t * th = lv_theme_mono_init(lv_disp_get_default(), false, LV_FONT_DEFAULT);
lv_disp_set_theme(lv_disp_get_default(), th);

/*Change the active screen's background color*/
lv_obj_set_style_bg_color(lv_screen_active(), lv_color_white(), LV_PART_MAIN);
/*Create a white label, set its text and align it to the center*/
lab = lv_label_create(lv_screen_active());
lv_obj_set_style_text_color(lv_screen_active(), lv_color_black(), LV_PART_MAIN);

while(1)
{
	lv_label_set_text(lab, "Hello World");
	lv_obj_align(lab, LV_ALIGN_CENTER, 0, 0);
	APPHAL_Delay(5000);
}
```

Any hints where I should be looking for?

mmurty, 2025-05-10:
I have had some progress over this issue. It seems I was successfully able to implement the flush callback. 
Following are the results on my LCD.

[image]

[image]

The pixels seems to be very large on the display so I don't feel there is something LVGL can do.
If I can improve the quality of the text somehow please suggest. (Like changing the font, or some settings in LVGL).
Finally if you feel these results are correct, I would like to contribute the LCD driver and flush_cb() implementation to the LVGL project. 
Please guide what would be the correct way for it.

closed 2025-06-08 by lvgl-bot