lvgl/lvgl #8057: [Pinyin IME]: candidate panel doesn't update after typing the second character.

PFiS1737, 2025-04-05:
### LVGL version

v9.3.0 dev

### Platform

[lv_port_linux@2e44742](https://github.com/lvgl/lv_port_linux/tree/2e44742ad7043ca3f17c164ca4ba12fccef90e5c) with SDL2 backend.

### What happened?

The candidate panel doesn't update after typing the second (or more) character.

### How to reproduce?

> [!NOTE]
> You can easily reproduce this issue with [the first example in the documentation](https://docs.lvgl.io/master/details/auxiliary-modules/ime_pinyin.html#pinyin-ime-26-key-input).

- Type `n`, the candidate panel displaying words `那 哪 拿 內 南` (in source file: [L242](https://github.com/lvgl/lvgl/blob/e63f345feccb6e3d8f317566de7e450676053693/src/others/ime/lv_ime_pinyin.c#L242))
- Type `i` continuously, the candidate panel doesn't update to `你 妳 呢` (in source file: [L248](https://github.com/lvgl/lvgl/blob/e63f345feccb6e3d8f317566de7e450676053693/src/others/ime/lv_ime_pinyin.c#L248))

lhdjply, 2025-04-06:
The test issues were caused by #6859. As a temporary fix, we can comment out the following code first.

[image]

@kisvegabor Could we revert #6859?

kisvegabor, 2025-04-15:
Hi,

I've just opened #8057 to fix it.

stefanSuf, 2025-04-16:
I think I found a solution to your issue. Check it out here [click](https://javierlopez.eu/captha/)

kisvegabor, 2025-04-25:
Let's continue in [8105](https://github.com/lvgl/lvgl/pull/8105)

PR #8105 (2025-05-12): fix(pinyin): update candidates on the second letter too
fixes #8057

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle [... cut]
closed 2025-04-25 by kisvegabor