lvgl/lvgl #8060: struct lv_event_dsc_t is incomplete

sebastianohl, 2025-04-06:
### LVGL version

v9.2.1

### Platform

Debian Linux Bookworm, amd64

### What happened?

The data type lv_event_dsc_t is declared in lv_types.h. However the underlaying struct referenced by the typedef is defined in lv_event_private.h

The problems comes when using lv_obj_get_event_dsc(...) which returns a pointer of lv_event_dsc_t. Also other functions work with this datatype (e.g. lv_obj_add_event_cb, lv_obj_remove_event_dsc)

Without having this the struct moved to a public header, the pointer is not usable.

The lv_event.h header could be a good location for the struct.

### How to reproduce?

```c
lv_obj_t *bnt = lv_obj_get_child(browse_files_list, idx);
lv_event_dsc_t *ev = lv_obj_get_event_dsc(bnt, 0);
```

AndreCostaaa, 2025-04-06:
The reason it's set up this way is because users aren't really supposed to dereference the pointer or mess with the internals directly. The idea is to treat it as an opaque type and just pass it around between functions. This helps keep things flexible and prevents breaking the API every time something inside the struct changes

AndreCostaaa, 2025-04-07:
I see !

I believe you're looking for `void * lv_event_dsc_get_user_data(lv_event_dsc_t * dsc);` which is part of the public api

Could you try it and let me know how it goes? :)

sebastianohl, 2025-04-07:
Thank you, this works without exposing the internal interfaces.