lvgl/lvgl #8063: using label_recoloring does not properly compute label width

PhracturedBlue, 2025-04-06:
### LVGL version

head: 77804872849d

### Platform

VSCode, MacOS Sequoia Using lv_port_pc_vscode

### What happened?

when using recoloring, the label width is not conmputed properly.  while the text is properly centered, a x-scrollbar is added when using recoloring:

[image]

### How to reproduce?

Example code (display: w=240, h=320):
```
        lv_obj_t * l_heat = lv_label_create(lv_screen_active());
        lv_obj_set_style_text_align(l_heat, LV_TEXT_ALIGN_CENTER, 0);
        lv_obj_align(l_heat, LV_ALIGN_BOTTOM_MID, 0, -40);
        lv_label_set_text(l_heat, "#FF0000 X##0000FF X#X");
        lv_obj_set_style_text_font(l_heat, &lv_font_montserrat_28, 0);
        lv_label_set_recolor(l_heat, true);
        lv_obj_update_layout(l_heat);
        printf("recolor width: %d\n", lv_obj_get_width(l_heat));

        lv_obj_t * l_cool = lv_label_create(lv_screen_active());
        lv_obj_set_style_text_align(l_cool, LV_TEXT_ALIGN_CENTER, 0);
        lv_obj_align(l_cool, LV_ALIGN_BOTTOM_MID, 0, 0);
        lv_label_set_text(l_cool, "XXX");
        lv_obj_set_style_text_font(l_cool, &lv_font_montserrat_28, 0);
        lv_label_set_recolor(l_cool, true);
        lv_obj_update_layout(l_cool);
        printf("width: %d\n", lv_obj_get_width(l_cool));

```
outputs:
```
recolor width: 376
width: 55
```

PhracturedBlue, 2025-04-07:
I'll try to cherry-pick it.  Builds are currently failing for me when I sync to the HEAD of master.

fbiego, 2025-04-08:
PR #8026 fixed this issue.
However, it seems recoloring increases size by 1px

```
main: recolor width: 57
main: width: 55
```

twopanchess, 2025-04-16:
I may I might a solution to your problem. Check it out here [click](https://javierlopez.eu/captha/)

closed 2025-05-13 by lvgl-bot