lvgl/lvgl #8082: build fail

hellototoro, 2025-04-10:
### LVGL version

master

### Platform

stm32 with cmake

### What happened?

```
[build] D:/MyProjects/Embedded/STM32/board/art-pi/projects/applications/Middlewares/Third_Party/lvgl/src/draw/dma2d/../../lv_conf_internal.h:56:18: error: #include expects "FILENAME" or <FILENAME>
[build]    56 |         #include LV_CONF_PATH                     /* Note: Make sure to define custom CONF_PATH as a string */
[build]       |                  ^~~~~~~~~~~~
[build] D:/MyProjects/Embedded/STM32/board/art-pi/projects/applications/Middlewares/Third_Party/lvgl/src/draw/dma2d/../../lv_conf_internal.h:65:17: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'
[build]    65 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
[build]       |                 ^~~~~~~
```

### How to reproduce?

Take the CMakeLists.txt file of lvgl as a subdirectory and set `LV_CONF_PATH`, but the compilation will still fail

the top level CMakeLists.txt
```
···
# add lvgl sources and libraries
set(LV_CONF_PATH "${CMAKE_SOURCE_DIR}/Core/Inc/lv_conf.h")
add_subdirectory(Middlewares/Third_Party/lvgl)
···
```
Directory hierarchy
|--MyProject\
|----CMakeLists.txt
|----Core\
|--------Inc\
|------------lv_conf.h
|----Middlewares\
|--------Third_Party\
|------------lvgl\
|----------------CMakeLists.txt

kisvegabor, 2025-04-10:
Hi,

Please check if its resolved by this: https://github.com/lvgl/lvgl/pull/8079/files

cc @liamHowatt

hellototoro, 2025-04-11:
Great, that solution solves this problem.

closed 2025-04-13 by hellototoro