lvgl/lvgl #8099: Display callback uint8_t px_map wrong color order

Schoumi, 2025-04-14:
### LVGL version

v9.2.2

### Platform

Esp-idf esp32

### What happened?

I implement a driver for a display I own. In the implementation I assume that the color received are RGB888 and it's what it's configured.

So I do not do anything to the data received in the display callback except sending them to the display driver where data are manipulated to be send to the hardware.
The problem is that RGB data turns to be BGR when I received them.

I'm not sure the problem is there but when I looked at lv_color_t I've seen that the order there is blue, green, red. lv_color_hex(0xff0000) store perfectly data in blue = 0, green = 0 and red = 255
but if  I read data from a pointer
```c
lv_color_t color = lv_color_hex(0xff0000);
uint8_t* ptr = (uint8_t*)&color;                                                                                  
printf("color: %d %d %d\n", ptr[0], ptr[1], ptr[2]);
```

then I get: `color: 0 0 255` which is correct according to the data structure. 

But if the px_map in the display callback is in fact a pointer to an array of lv_color_t, then the data are not store as RGB but BGR.

### How to reproduce?

```c
//somewhere set this
lv_obj_set_style_bg_color(active_screen, lv_color_hex(0xff0000), LV_PART_MAIN);

//in the display callback
printf("color: %d %d %d\n", px_map[0], px_map[1], px_map[2]);
```

liamHowatt, 2025-04-15:
I may be wrong but I think that's how the colors are typically ordered in buffers. Most displays expect this order.

Schoumi, 2025-04-15:
To me it's very weird since everybody talk about rgb so i expect things to be in rgb order not bgr

uLipe, 2025-04-16:
I think we can close this one, since @Schoumi will do the inversion for his case.

Feel free to reopen

closed 2025-04-16 by uLipe