lvgl/lvgl #8108: ffmpeg `lv_image_set_src()` broken by PR #7969

AndyEveritt, 2025-04-15:
### LVGL version

v9.2.0-748-g8da142549

### Platform

WSL

### What happened?

Running this code:

```cpp
lv_obj_t* image = lv_image_create(cont);
lv_image_set_src(image, "A:assets/ffmpeg.png");
```

With `LV_USE_FFMPEG=1` and `LV_USE_LIBPNG=0` in `lv_conf.h`

The code fails the assertion added by PR #7969 in `lv_image_decoder_open()`

```
/*
     * We assume that if a decoder can get the info, it can open the image.
     * If decoder open failed, free the source and return error.
     * If decoder open succeed, add the image to cache if enabled.
     * */
    lv_result_t res = dsc->decoder->open_cb(dsc->decoder, dsc);

    if(res == LV_RESULT_OK && dsc->decoded != NULL) {
        LV_ASSERT_MSG(dsc->decoded->unaligned_data && dsc->decoded->handlers, "Invalid draw buffer");

        /* Flush the D-Cache if enabled and the image was successfully opened */
        if(dsc->args.flush_cache) {
            lv_draw_buf_flush_cache(dsc->decoded, NULL);
            LV_LOG_INFO("Flushed D-cache: src %p (%s) (W%d x H%d, data: %p cf: %d)",
                        src,
                        dsc->src_type == LV_IMAGE_SRC_FILE ? (const char *)src : "c-array",
                        dsc->decoded->header.w,
                        dsc->decoded->header.h,
                        (void *)dsc->decoded->data,
                        dsc->decoded->header.cf);
        }
    }

    return res;
```

`LV_ASSERT_MSG(dsc->decoded->unaligned_data && dsc->decoded->handlers, "Invalid draw buffer");` fails because `unaligned_data` is set to `NULL` by ffmpeg in `decoder_open()`. `handlers` is also `NULL`

If this assert is removed then the image is displayed correctly. Otherwise the code enters an infinitely while loop.

emomyname, 2025-04-16:
I may I might a solution to your issue. Check it out here [click](https://javierlopez.eu/captha/)

FASTSHIFT, 2025-04-18:
fix here: https://github.com/lvgl/lvgl/pull/8128

closed 2025-04-23 by kisvegabor