lvgl/lvgl #8121: Typedef

ImolaS3, 2025-04-16:
### LVGL version

v9.xx

### Platform

Arduino IDE for ESP32 board and amoLED display.

### What happened?

I am new to this library, and my code works in v8.4, but when installing v9.3, it is ridiculous. nothing but errors where constants and functions have been renamed. these are navigable (at the waste of a lot of time), but i also see a Typedef error quite often
Please sort this out. V9.xx appears to have been. around a while, yet it also appears to be very buggy, particularly with the typedef errors which a user should not have to investigate when migrating
 

### How to reproduce?

just try to use v9 with exisiting code and examples from lilygo or the arduino IDE

uLipe, 2025-04-16:
Hello!

Thanks for reporting that, could you share a snippet of the error console?

Thank you

C47D, 2025-04-27:
Hi,

I usually get those kind of errors after updating my local repository when using the Eclipse simulator, and solve them by running the project index, I don't know if that's available in the arduino ide.

closed 2025-05-20 by lvgl-bot