lvgl/lvgl #8123: A segmentation fault occurs when using LV_OS_PTHREAD

makun666, 2025-04-17:
### LVGL version

v9.2

### Platform

Linux embedded system

### What happened?

When I enable multi-threaded rendering by setting LV_USE_OS to LV_OS_PTHREAD, a segmentation fault occurs.

The key configurations are as follows:
LV_USE_OS:LV_OS_PTHREAD
LV_DRAW_SW_DRAW_UNIT_CNT:2 (my platform's CPU has two cores)
LV_DRAW_THREAD_STACK_SIZE    (32 * 1024) 
LV_CACHE_DEF_SIZE : (1010241024).

In most simple UI scenarios, the program runs normally, and the UI displays correctly. 
However, in some complex scenarios, such as when multiple images or animations need to be loaded, 
a segmentation fault inevitably occurs. 
Sometimes, it also reports an invalid pointer error.

The backtrace when the segmentation fault occurs is as follows:
(gdb) bt
#0  0x004824b0 in lv_cache_entry_get_ref ()
#1  0x004820a8 in get_victim_cb ()
#2  0x004807b8 in cache_evict_one_internal_no_lock ()
#3  0x004808e8 in cache_add_internal_no_lock ()
#4  0x00480e14 in lv_cache_add ()
#5  0x00465edc in lv_image_decoder_add_to_cache ()

Different backtraces sometimes:
(gdb)
#0  0x0045ea90 in lv_draw_buf_destroy ()
#1  0x00482a74 in image_cache_free_cb ()
#2  0x00480838 in cache_evict_one_internal_no_lock ()
#3  0x004808e8 in cache_add_internal_no_lock ()
#4  0x00480e14 in lv_cache_add ()
#5  0x00465edc in lv_image_decoder_add_to_cache ()

When I set LV_USE_OS to LV_OS_NONE, all operations run normally without this issue.

kisvegabor, 2025-04-29:
@W-Mai do you have any idea?

W-Mai, 2025-04-29:
> [@W-Mai](https://github.com/W-Mai) do you have any idea?

I think this is indeed quite difficult to solve at the moment. I'm not sure if it's just a caching issue, because LVGL actually doesn't lock many places, or uses atomic counters.
It's the same with this:

- https://github.com/lvgl/lvgl/pull/6665

W-Mai, 2025-04-29:
I think, during the upcoming Labor Day holiday, I can take a look at this issue. I'm not sure if I have any ideas to solve it 😖.

makun666, 2025-05-06:
I added some logs in the cache-related code. After reviewing the code and the logs, here's my analysis:

In the lv_image_decoder_open function, it first calls try_cache. If there's a cache hit, it directly returns LV_RESULT_OK; otherwise, it sequentially calls open_cb -> lv_image_decoder_add_to_cache -> lv_cache_add to add the decoded data to the cache.

Here are the problematic logs observed when LV_OS_PTHREAD is enabled:

[image]

Looking at the logs, it's found that the same src is called twice consecutively in lv_image_decoder_open. In a single-core, serial execution scenario, during the first call, try_cache misses, and then the data is added to the cache. During the second call, try_cache hits, and the function returns successfully. The single-core operation works normally.

However, when LV_OS_PTHREAD is enabled and running on a dual-core system, both try_cache calls miss. The same src triggers two calls to lv_cache_add -> cache_add_internal_no_lock -> add_cb -> alloc_new_node.

In lv_cache_add, a mutex is used for lock protection. When calling lv_rb_insert in alloc_new_node, there's a check to see if the node already exists. But when the node already exists in the RB tree, it still directly calls lv_ll_ins_head to insert a new node into the linked list. Is there an issue with this logic?

I tried adding a check in cache_add_internal_no_lock to see if the cache entry already exists. After this addition, the segment fault problem no longer occurs, but there's a memory leak issue.

Below is the code I added:

[image]

I hope this analysis can assist you in investigating this problem.

makun666, 2025-05-12:
@W-Mai Has there been any progress in the investigation into this issue? 
Just let me know if you need me to assist with any tests

makun666, 2025-05-15:
The cause of the memory leak has been identified. It appears that when the same resource misses in the try_cache for all attempts, multiple decoding processes are initiated through the decoder, resulting in the generation of two copies of decoded data. When deleting the cache, only one copy of the decoded data is released, leading to the memory leak.

W-Mai, 2025-05-15:
> The cause of the memory leak has been identified. It appears that when the same resource misses in the try_cache for all attempts, multiple decoding processes are initiated through the decoder, [...]

So, could it be that it's the decoder itself that's not handling multi-threading properly?
What do you think?

makun666, 2025-05-16:
> So, could it be that it's the decoder itself that's not handling multi-threading properly? What do you think?

Yes, in a situation like this, it seems that the only way to avoid problems is to protect the entire try_cache/decoder_open/add_cache process with a mutex.

closed 2025-07-05 by lvgl-bot

AndreCostaaa, 2025-07-17:
Hi @makun666 thank you for your investigation and i'm sorry this has gone stale. A similar issue #8542 was recently opened and i'm labeling it as important so it doesn't get closed as well, we need to allocate some time to investigate and fix it