lvgl/lvgl #8124: it occurs wihte spot when i set the hight lower than 30. My screen size is 240(H)*280(V)

Simple-Young, 2025-04-17:
### LVGL version

v9.2.2

### Platform

platform: esp32s3 + st7789(lcd driver)

### What happened?

[image]

when i set the hight < 30, the white spot occurs.
it seems the Rendering Engine's duty.

### How to reproduce?

the code:
```c
    this.scr = lv_scr_act();
    lvgl_port_lock(0);
    lv_obj_set_style_bg_color(this.scr, ATZ_LV_COLOR_GREEN, 0);

    lv_obj_t* mouth = lv_obj_create(this.scr);
    lv_obj_set_size(mouth, 50, 25);
    lv_obj_set_align(mouth, LV_ALIGN_CENTER);

    lv_obj_set_style_radius(mouth, 0, 0);
    lv_obj_set_style_bg_color(mouth, ATZ_LV_COLOR_RED, 0);
    lv_obj_set_style_bg_opa(mouth, LV_OPA_COVER, 0);

    lv_obj_set_style_border_width(mouth, 0, 0);
    lv_obj_set_style_border_color(mouth, ATZ_LV_COLOR_GREEN, 0);
    lv_obj_set_style_border_opa(mouth, LV_OPA_COVER, 0);

    lv_obj_set_style_shadow_width(mouth, 0, 0);
    lv_obj_set_style_shadow_color(mouth, ATZ_LV_COLOR_GREEN, 0);
    lv_obj_set_style_shadow_opa(mouth, LV_OPA_COVER, 0);
    this.robot.mouth = mouth;

    lvgl_port_unlock();
```

Simple-Young, 2025-04-17:
---------------------------------the new one----------------------------------------------
i try the same code in windows simulator, it doesn't occurs.

FASTSHIFT, 2025-04-18:
This is [scrollbar](https://docs.lvgl.io/master/details/common-widget-features/scrolling.html#scrollbar), you can disable it.

closed 2025-04-21 by Simple-Young