lvgl/lvgl #8127: Reduce size of test directory

AndyEveritt, 2025-04-17:
### Introduce the problem

The current test directory is ~25GB once tests are built.

I believe this is because the lvgl library is compiled as a static library instead of a dynamic library. This means that each test executable is ~60MB.

### Proposal

If the lvgl library is compiled dynamically then the test folder should be substaintially smaller in size.

By adding `set(BUILD_SHARED_LIBS ON CACHE BOOL "Build shared libraries")` in `tests/CMakeLists.txt`, lvgl can be set to be built as a dynamic library.

However when running `cmake --build /home/lvgl/tests/build_test_defheap` (called by `python tests/main.py build test`), the following error occurs.

```
==================================================================
Building: test_defheap: Test config, LVGL heap, 32 bit color depth
==================================================================

[6/595] Linking C executable test_cache
FAILED: test_cache 
: && /usr/bin/cc -g  CMakeFiles/test_cache.dir/src/test_cases/cache/test_cache.c.o CMakeFiles/test_cache.dir/test_cache_Runner.c.o CMakeFiles/test_cache.dir/wayland_protocols/wayland_xdg_shell.c.o -o test_cache  -Wl,-rpath,/home/lvgl/tests/build_test_defheap/lib  -lwayland-client  -lwayland-cursor  -lxkbcommon  libtest_common.a  lib/liblvgl_demos.so.9.3.0  lib/liblvgl_thorvg.so.9.3.0  /usr/lib/x86_64-linux-gnu/libpng.so  /usr/lib/x86_64-linux-gnu/libz.so  /usr/lib/x86_64-linux-gnu/libfreetype.so  /usr/lib/x86_64-linux-gnu/libjpeg.so  -lm  -lpthread  -fsanitize=address  -fsanitize=leak  -fsanitize=undefined  --coverage  libtest_libs.a  lib/liblvgl.so.9.3.0  -lwayland-client  -lwayland-cursor  -lxkbcommon && :
/usr/bin/ld: test_cache: hidden symbol `__gcov_init' in /usr/lib/gcc/x86_64-linux-gnu/11/libgcov.a(_gcov.o) is referenced by DSO
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
[7/595] Linking C executable test_render_to_argb2222
FAILED: test_render_to_argb2222 
: && /usr/bin/cc -g  CMakeFiles/test_render_to_argb2222.dir/src/test_cases/draw/test_render_to_argb2222.c.o CMakeFiles/test_render_to_argb2222.dir/test_render_to_argb2222_Runner.c.o CMakeFiles/test_render_to_argb2222.dir/wayland_protocols/wayland_xdg_shell.c.o -o test_render_to_argb2222  -Wl,-rpath,/home/lvgl/tests/build_test_defheap/lib  -lwayland-client  -lwayland-cursor  -lxkbcommon  libtest_common.a  lib/liblvgl_demos.so.9.3.0  lib/liblvgl_thorvg.so.9.3.0  /usr/lib/x86_64-linux-gnu/libpng.so  /usr/lib/x86_64-linux-gnu/libz.so  /usr/lib/x86_64-linux-gnu/libfreetype.so  /usr/lib/x86_64-linux-gnu/libjpeg.so  -lm  -lpthread  -fsanitize=address  -fsanitize=leak  -fsanitize=undefined  --coverage  libtest_libs.a  lib/liblvgl.so.9.3.0  -lwayland-client  -lwayland-cursor  -lxkbcommon && :
/usr/bin/ld: test_render_to_argb2222: hidden symbol `__gcov_init' in /usr/lib/gcc/x86_64-linux-gnu/11/libgcov.a(_gcov.o) is referenced by DSO
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status

...

ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/home/lvgl/tests/main.py", line 262, in <module>
    build_tests(options_name, build_type, args.clean)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lvgl/tests/main.py", line 138, in build_tests
    subprocess.check_call(['cmake', '--build', build_dir,
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                           '--parallel', str(os.cpu_count())])
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/subprocess.py", line 421, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '/home/lvgl/tests/build_test_defheap', '--parallel', '20']' returned non-zero exit status 1.
```

It appears that making the library dynamic has caused an issue with the coverage library.

Is there an easy solution to this?

AndyEveritt, 2025-04-17:
The `lvgl/tests` folder
[image]

[image]

[image]

KyL0N, 2025-04-24:
can't agree more

AndyEveritt, 2025-05-08:
Not stale

MattCatz, 2025-05-09:
I looked at this for a little bit. Maybe someone who understands the cmake system may be able to push it further than me.

The problems seen when building as a shared library mostly stem from dependency propagation. For example the `__gcov_init` problem is because the shared library version is not built with the `--coverage` flag. This can be verified by looking in the `build.ninja` file for the flags:
```
#############################################
# Link the shared library lib/liblvgl.so.9.3.0

build lib/liblvgl.so.9.3.0: C_SHARED_LIBRARY_LINKER__lvgl_Debug CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_group.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_class.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_draw.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_event.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_id_builtin.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_pos.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_property.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_scroll.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_style.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_style_gen.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_obj_tree.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/core/lv_refr.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/display/lv_display.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/dma2d/lv_draw_dma2d.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/dma2d/lv_draw_dma2d_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/dma2d/lv_draw_dma2d_img.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_3d.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_arc.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_buf.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_image.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_label.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_line.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_mask.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_rect.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_triangle.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_draw_vector.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/lv_image_decoder.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_arc.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_border.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_img.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_label.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_layer.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_line.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_stm32_hal.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_triangle.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_draw_nema_gfx_utils.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nema_gfx/lv_nema_gfx_path.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_draw_buf_g2d.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_draw_g2d.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_draw_g2d_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_draw_g2d_img.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_g2d_buf_map.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/g2d/lv_g2d_utils.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_draw_buf_pxp.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_draw_pxp.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_draw_pxp_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_draw_pxp_img.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_draw_pxp_layer.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_pxp_cfg.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_pxp_osa.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/pxp/lv_pxp_utils.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_buf_vglite.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_arc.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_border.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_img.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_label.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_layer.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_line.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_draw_vglite_triangle.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_vglite_buf.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_vglite_matrix.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_vglite_path.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/nxp/vglite/lv_vglite_utils.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/opengles/lv_draw_opengles.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d_arc.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d_border.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d_fill.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d_image.c.o CMakeFiles/lvgl.dir/tmp/lvgl/src/draw/renesas/dave2d/lv_draw_dave2d_label. [... cut ...]
  POST_BUILD = :
  PRE_LINK = :
  SONAME = liblvgl.so.9
  SONAME_FLAG = -Wl,-soname,
  TARGET_COMPILE_PDB = CMakeFiles/lvgl.dir/
  TARGET_FILE = lib/liblvgl.so.9.3.0
  TARGET_IMPLIB = lib/
  TARGET_PDB = lib/liblvgl.pdb
```

Ading something crude like `set(CMAKE_SHARED_LINKER_FLAGS "${CMAKE_SHARED_LINKER_FLAGS} --coverage")` should fix the `__gcov_init` problem but then it's playing whack-a-mole with all the other libraries that seem to be implicitly linked with the static version (like libdrm).

MattCatz, 2025-05-09:
I was able to get things working by adding `link_libraries(...)` in a few places (e.x. `link_libraries(drm)`). On my machine there is about a 95% reduction in size between the static library and shared library tests. Results below:

## Static Tests
```
real	6m6.152s
user	49m1.108s
sys	7m22.393s
```
```
25G	./tests/
```
```
36M	./tests/build_16bit
82M	./tests/build_24bit
82M	./tests/build_full_32bit
36M	./tests/build_normal_8bit
83M	./tests/build_sdl
7.7G	./tests/build_test_defheap
7.9G	./tests/build_test_sysheap
8.4G	./tests/build_test_vg_lite
86M	./tests/build_vg_lite
20K	./tests/CMakeLists.txt
4.0K	./tests/config.yml
4.0K	./tests/FindLibDRM.cmake
4.0K	./tests/fs_read_random.bin
4.0K	./tests/fs_write_read_random.bin
8.0K	./tests/gen_json
12K	./tests/main.py
16K	./tests/makefile
24K	./tests/makefile_uefi
312K	./tests/micropy_test
4.0K	./tests/README.md
18M	./tests/ref_imgs
19M	./tests/ref_imgs_vg_lite
19M	./tests/src
2.7M	./tests/test_images
316K	./tests/unity
96K	./tests/wayland_protocols
```

## Dynamic Tests
```
real	5m20.313s
user	44m35.149s
sys	5m43.040s
```
```
1.2G	./tests
```
```
29M	./tests/build_16bit
63M	./tests/build_24bit
63M	./tests/build_full_32bit
29M	./tests/build_normal_8bit
63M	./tests/build_sdl
286M	./tests/build_test_defheap
280M	./tests/build_test_sysheap
292M	./tests/build_test_vg_lite
66M	./tests/build_vg_lite
20K	./tests/CMakeLists.txt
4.0K	./tests/config.yml
4.0K	./tests/FindLibDRM.cmake
4.0K	./tests/fs_read_random.bin
4.0K	./tests/fs_write_read_random.bin
8.0K	./tests/gen_json
12K	./tests/main.py
16K	./tests/makefile
24K	./tests/makefile_uefi
312K	./tests/micropy_test
4.0K	./tests/README.md
18M	./tests/ref_imgs
19M	./tests/ref_imgs_vg_lite
19M	./tests/src
2.7M	./tests/test_images
316K	./tests/unity
96K	./tests/wayland_protocols
```