lvgl/lvgl #8130: Custom font not displaying in clarity

keerthansys22, 2025-04-18:
### LVGL version

v9.2.1

### Platform

Linux environment

### What happened?

I’m trying to display the custom font - Monospace821 BT in lvgl by taking the ttf file and converting that in lvgl font convertor to source file.

Earlier in V8.7, there was an option called subpixel rendering in lv_conf.h, but in this v9.2.1 we don't have that. 

But there is no crystal clarity in fonts display, some fonts are not clear it is bleeding! How to resolve this? Kindly help, thanks!

NehalAR, 2025-04-29:
@keerthansys22 Can you share the actual ttf file, code file how are you displaying and some pictures how its looking on your screen? It would help to narrow down the problem. Also if possible share what fonts are displaying properly, are they all the build in ones?

keerthansys22, 2025-04-30:
Hi @NehalAR , thanks for your message. My requirement is to use Monospace821 BT. The fonts I'm using is not a built-in-ones, it is a custom font, and I got the source file from ttf via lvgl font convertor.

TTF source for Monospace821 BT font is - https://www.cufonfonts.com/font/monospac821-bt (I have used the fonts from different sources from web too but not improvements in clarity)

I'm attaching the three snapshots, where the fonts are not clear to read. In first and third image, the string "TBC" to be displayed but it is displaying as "TBG". "/" is not displayed properly!

[image]
[image]
[image]

The configuration used in lvgl font convertor is 1 BPP, sub-pixel rendering can't be used since the configuration to enable that in lvgl V9.1 is not provided.

NehalAR, 2025-04-30:
I did convert the font with the recommend converter and here's what I used the settings:
```

/*******************************************************************************
 * Size: 72 px
 * Bpp: 4
 * Opts: --bpp 4 --size 72 --no-compress --font Montserrat-Medium.ttf --range 32-127,176,8226 --format lvgl -o LV_FONT_MONTSERRAT_72.c
 ******************************************************************************/
```

Even if you look into the actual font .c files in _lvgl/src/font/_ you can find what settings are used which I also set as I wasn't getting correct fonts drawn either.

keerthansys22, 2025-05-06:
Hi @NehalAR, any suggestions or any alternative methods you tried experimenting with to enhance font clarity. If yes, let me know even I will experiment on my side. Thanks.

keerthansys22, 2025-05-21:
I want to make sure that I'm adding custom fonts in lv_conf.h properly. 

Adding the #define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(monospace_12) LV_FONT_DECLARE(monospace_11) and taking these fonts in the application use. Is this right know?

closed 2025-07-05 by lvgl-bot