lvgl/lvgl #8147: How to create circle chart?

IAMMX, 2025-04-23:
### Introduce the problem

Currently, lv_chart doesn't support LV_CHART_TYPE_CIRCLE. Could we add this feature? 

[image]

[image]

lizhaoming634, 2025-04-23:
in lvgl8, you can use lv_meter
https://docs.lvgl.io/8.3/widgets/extra/meter.html#pie-chart

IAMMX, 2025-04-24:
> in lvgl8, you can use lv_meter https://docs.lvgl.io/8.3/widgets/extra/meter.html#pie-chart

v9.2.2

lizhaoming634, 2025-04-24:
you can use lv_scale,
https://github.com/lvgl/lvgl/tree/master/demos/widgets

[image]

kisvegabor, 2025-04-29:
You can use `lv_arc`.

closed 2025-05-10 by IAMMX