lvgl/lvgl #8151: Parentesis missing in lv_conf.cmake, compiling error

lukilukeskywalker, 2025-04-23:
### LVGL version

v9.3.0-dev commit: ce69f4c 

### Platform

Linux Ubuntu, compile inside a Docker container, although the emulation is being done via SDL2 in the host system. A Makefile lets me choose for what target I want to build the firmware

[Dockerfile.txt](https://github.com/user-attachments/files/19868741/Dockerfile.txt)

[CMakeLists.txt](https://github.com/user-attachments/files/19868907/CMakeLists.txt)

[simulation.txt](https://github.com/user-attachments/files/19869012/simulation.txt)

### What happened?

When trying to compile with CMake, CMake creates a faulty lv_conf.cmake, where a single ")" is missing, in my case line 141: set(CONFIG_LV_USE_GENERIC_MIPI (LV_USE_ST7735) Which, makes the whole compilation fail. 
In the v9.3.0-dev commit: [lvgl @ 7780487](https://github.com/lvgl/lvgl/tree/77804872849d4119fecc83d70697626de00b47c5) the compilation did still work without any hitch

[image]

### How to reproduce?

Not sure about that, I am not sure if it is my fault or something is wrong in the new lvgl commits

lukilukeskywalker, 2025-04-24:
Also, I have another question, why does ce69f4c need a python env and 7780487 doesn't? What has changed in the build system?

AndreCostaaa, 2025-04-24:
Hi there,

Check out this answer for the fix

https://github.com/lvgl/lvgl/issues/8046#issuecomment-2777925788

As to what happened,

LVGL added a new build dependency in order to add native kconfig support through cmake #7934

With that change the previous way GENERIC_MIPI was defined no longer works and needs to be updated (check the link above with the fix)

Eventually the pcpp dependency was removed and replaced with python-venv as it's a common enough dependency that shouldn't cause too much problems for users #8090

lukilukeskywalker, 2025-04-24:
Yeah, I just wanted to say that I just saw the issue: https://github.com/lvgl/lvgl/issues/8046 with your comment, and I just tested it out, and now it works... Somehow I did miss this change when I did a diff from the new lv_conf_template and the old lv_conf
I am sorry and thank you very much!!!