lvgl/lvgl #8152: Lag if there is a lot of touchscreen events

ithinu, 2025-04-23:
### Introduce the problem

Is LVGL consuming all available touch events in a single cycle? I have a FIFO between the touchscreen driver and `touchpad_read(lv_indev_t * indev, lv_indev_data_t * data)` and if the cursor moves fast, the FIFO fills up quickly, causing a lag much longer than `LV_DEF_REFR_PERIOD`.

The platform is Pico 2 + cst816s + ST7789T3. I wrote the driver for cst816s, maybe it is buggy but logs show that what is pushed to the FIFO looks reasonable.

kisvegabor, 2025-04-29:
Hey,

In your `read_cb` you can set `data->continue_reading = true;` if the FIFO is not empty, so that LVGL will call the `read_cb` again.

arturrataj, 2025-04-29:
Works, thanks.

closed 2025-04-30 by ithinu