lvgl/lvgl #8159: LV_COLOR_FORMAT_RGB565A8 is not SUPPORTED for lv_canvas_set_buffer

linghuye, 2025-04-25:
### Introduce the problem

I use LV_COLOR_DEPTH 16 on esp32-c6. When I try to use a transparent canvas widget, I call lv_canvas_set_buffer with LV_COLOR_FORMAT_NATIVE_WITH_ALPHA which is LV_COLOR_FORMAT_RGB565A8, and finally found it is not supported for draw buffer. So I have to use LV_COLOR_FORMAT_ARGB8888.
It is strange that LV_COLOR_FORMAT_NATIVE_WITH_ALPHA is not supported for render target and not documented anywhere, 
```
 LV_COLOR_FORMAT_ARGB8565          = 0x13,   /**< Not supported by sw renderer yet. */
 LV_COLOR_FORMAT_RGB565A8          = 0x14,   /**< Color array followed by Alpha array*/ It is also NOT SUPPORTED
```
So there is no way to render with alpha except LV_COLOR_FORMAT_ARGB8888, am I correct or I misunderstand something?
And lv_color_format_get_bpp(LV_COLOR_FORMAT_RGB565A8) return 16?``

kisvegabor, 2025-04-29:
That's correct, LVGL RGB565A8 can be source color format, but it can not be a destination color format. Meaning LVGL can not render in this color format. So it's supported by the SW renderer, but only as source. 

I suggest using ARGB8888. 
 
> And lv_color_format_get_bpp(LV_COLOR_FORMAT_RGB565A8) return 16?

I started this sentence like 5 times to convince you and myself that it better that it's 16 and not 24, but actually it really should be 24... Basically we `if`s everywhere to support this special color format  to actual value is used only on a few places.

linghuye, 2025-05-01:
Thanks for your clarification. I understand the design choice now.

closed 2025-05-01 by linghuye