lvgl/lvgl #8170: How to build in offline env: pip install pcpp in python venv

fzboffice, 2025-04-28:
### LVGL version

v9.3.0

### Platform

any platform

### What happened?

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f125cc19550>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/pcpp/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f125cc19e20>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/pcpp/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f125cc0cf70>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/pcpp/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f125cc193d0>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/pcpp/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f125cc19220>: Failed to establish a new connection: [Errno 101] Network is unreachable')': /simple/pcpp/
ERROR: Could not find a version that satisfies the requirement pcpp (from versions: none)
ERROR: No matching distribution found for pcpp

### How to reproduce?

I have made some attempts and am able to use it temporarily:
I change line 104 of preprocess_lv_conf_internal.py to: subprocess.check_call([venv_pip, "install", "--no-index", "--find-links=~/pcpp", "pcpp"])

AndreCostaaa, 2025-04-28:
Hello,

Thank you for opening this issue. Building LVGL offline is something we didn't really think through

Good news tho: with #8126 merged, `pcpp` is no longer used by default and your build issues should now be gone.
Please try updating to the latest master branch and let me know how if it works

We'll be looking into re-enabling `pcpp` back in the future but will consider this issue and a way to circumvent it

fzboffice, 2025-04-29:
@AndreCostaaa thank you,well done.it works well.

closed 2025-04-29 by fzboffice