lvgl/lvgl #8176: Application crush on invalid column row clicked on lv_file_explorer.c

GS-x, 2025-04-29:
### LVGL version

master

### Platform

rk3566 - wayland backend

### What happened?

The application crush when we get a NULL file from the line 507 on lv_file_explorer.c

### How to reproduce?

master brunch of lv_file_explorer.c

C47D, 2025-04-30:
Which of the 3 examples are you using? I've tried all of them and there's no empty column to click in, but I agree we should check if `lv_table_get_cell_user_data` returns NULL.

GS-x, 2025-04-30:
I started from lv_example_file_explorer_1.c and imported the example in my project.
During the file browsing, I just clicked in the empty area (under the list of files and directories). This event return a NULL value because it's outside the table area created by the file explorer.

C47D, 2025-05-01:
Hi,

This patch should avoid the crashing, can you try it out?
```diff
diff --git a/src/others/file_explorer/lv_file_explorer.c b/src/others/file_explorer/lv_file_explorer.c
index 0563aaeef..1c91d29cc 100644
--- a/src/others/file_explorer/lv_file_explorer.c
+++ b/src/others/file_explorer/lv_file_explorer.c
@@ -505,6 +505,10 @@ static void browser_file_event_handler(lv_event_t * e)
         lv_table_get_selected_cell(explorer->file_table, &row, &col);
         selected_text = lv_table_get_cell_value(explorer->file_table, row, col);
         file_entry_user_data = lv_table_get_cell_user_data(explorer->file_table, row, col);
+        if(file_entry_user_data == NULL) {
+            LV_LOG_USER("No user data on selected entry");
+            return;
+        }
 
         selected_text = selected_text + 5; /* skip table cell format */
 

```

GS-x, 2025-05-07:
It seams that works. Thank you.

liamHowatt, 2025-05-08:
@C47D please open a PR for it 😁

closed 2025-06-01 by lvgl-bot