lvgl/lvgl #8191: Custom memory allocator with CONFIG_LV_USE_CUSTOM_MALLOC does not link correctly when defined outside LVGL component (ESP-IDF)

BatuhanKaratas, 2025-05-02:
### LVGL version

9.2.2

### Platform

ESP32-S3 with ESP-IDF v5.4.1

### What happened?

We enabled CONFIG_LV_USE_CUSTOM_MALLOC in the LVGL configuration to override LVGL's default memory allocation with our custom implementation using `heap_caps_malloc(size, MALLOC_CAP_SPIRAM)` `**to**` allocate all LVGL memory in PSRAM.

We followed the advice from [issue #7066](https://github.com/lvgl/lvgl/issues/7066) and implemented all necessary functions (e.g., `lv_malloc_core`, `lv_realloc_core`, `lv_free_core,` etc.) in our own `lv_malloc_core.c` file. However, as also noted by @giantbing in that issue, we encountered a link-time error: LVGL's static library did not link our versions of the allocation functions defined in the main application component.

To make it work, we had to modify the original `lv_mem_core_clib.c` in the LVGL source directory by replacing its function definitions with our own, as shown below. This worked, but it breaks the managed component model, since it requires modifying the contents of the lvgl library under `.managed_components`.

### How to reproduce?

We expected that defining lv_malloc_core and related functions in our application component (outside the LVGL source tree) would override the weak/default implementations during linking — without having to modify the contents of the managed LVGL component.

**How to reproduce?**

1. Use ESP-IDF v5.4.1 with LVGL v9.2.2 as a managed component.
2. Enable CONFIG_LV_USE_CUSTOM_MALLOC.
3. Define custom memory functions in your application component:

```
void* lv_malloc_core(size_t size) {
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
}

void* lv_realloc_core(void* p, size_t new_size) {
    return heap_caps_realloc(p, new_size, MALLOC_CAP_SPIRAM);
}

void lv_free_core(void* p) {
    heap_caps_free(p);
}

```
4. Build the project. You’ll see link-time errors indicating that your custom symbols were not used.
5. If you instead replace lv_mem_core_clib.c inside the LVGL source with the above code, it works — but this is not sustainable in a managed component environment.

**Proposed Solution**
We suggest marking the default memory allocation functions (e.g., lv_malloc_core, lv_realloc_core, lv_free_core, etc.) in `lv_mem_core_clib.c` with the` __attribute__((weak))` GCC attribute.
This would allow user applications to cleanly override these functions at link time without modifying LVGL source code.

```
__attribute__((weak)) void* lv_malloc_core(size_t size) {
    return malloc(size);
}
```
This is a common and effective pattern in embedded development, especially in frameworks intended for integration with user-defined environments.

Thank you!

BatuhanKaratas, 2025-05-27:
I got same @[giantbing](https://github.com/giantbing) link error message (`undefined reference`). Since I set a custom allocator, it doesn't give a multiple definition error.

xperiments, 2025-09-09:
Problem persist, why not planned? Is a HARD bug
Will be great if there is a place in the documentation where all of this is explained

AndreCostaaa, 2025-09-09:
Hi everyone, i can't reproduce your error. I noticed that you're missing the definition of `lv_mem_init` in your sample code so maybe that's why it's failing for you?

After setting `CONFIG_LV_USE_CUSTOM_MALLOC=y` in my `sdkconfig` :

```
# CONFIG_LV_USE_BUILTIN_MALLOC is not set
# CONFIG_LV_USE_CLIB_MALLOC is not set
# CONFIG_LV_USE_MICROPYTHON_MALLOC is not set
# CONFIG_LV_USE_RTTHREAD_MALLOC is not set
CONFIG_LV_USE_CUSTOM_MALLOC=y
```
And defining these in `main.c`

```c
void lv_mem_init(void)
{
}
void *lv_malloc_core(size_t size)
{
	return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);
}

void *lv_realloc_core(void *p, size_t new_size)
{
	return heap_caps_realloc(p, new_size, MALLOC_CAP_SPIRAM);
}

void lv_free_core(void *p)
{
	heap_caps_free(p);
}
```

`idf.py build` completes successfully. 

If you have a sample application that I can take a look let me know, until then I'm closing this issue. Thank you for your patience

closed 2025-09-09 by AndreCostaaa

dlugaz, 2026-09-22:
For anyone searching for a solution to this problem - https://forum.lvgl.io/t/how-to-implement-link-lvgl-custom-external-malloc-in-esp-idf/23723/2
It is caused by random linking order in a standard idf project configuration. Just reposting the key points from the OP.
The solution is to modify CMake files. 
in the component where your malloc implementation .c lies:
`idf_component_register(
SRCS “your_memory.c”
INCLUDE_DIRS “include”
// Forces the linker to include all the functions of this component.
// even if no one explicitly calls them in the main
WHOLE_ARCHIVE // I only added this to my idf_component_register because the rest I already had.
)

// This tells CMake: "Whoever uses me (or whoever the system links to)
// It needs to be linked to my memory functions.
target_link_libraries(${COMPONENT_LIB} INTERFACE “-u lv_malloc_core”)
target_link_libraries(${COMPONENT_LIB} INTERFACE “-u lv_realloc_core”)
target_link_libraries(${COMPONENT_LIB} INTERFACE “-u lv_free_core”)`

In the root directory CMakeLists.txt:
`idf_build_set_property(LINK_COMPONENTS “lv_mem_custom_alloc” APPEND)`