lvgl/lvgl #8202: Start and completed callbacks on animation timelines

lorenzo-arena, 2025-05-05:
### LVGL version

v9.2.0

### Platform

Custom SDL driver on Linux

### What happened?

I'm trying to use the animation timeline to chain multiple animations; I'd also like to use the `start_cb` of one of these animations (not the first one); however it only gets called at the beginning of the timeline. Is this expected behavior? I would have liked to have it being called every time the specific animation for which I set the callback starts (even more so if the timeline is looping).

### How to reproduce?

In the *lv_example_anim_timeline_1.c* example, add a start_cb on one of the animation which start later, like `a3` or `a4`.

lorenzo-arena, 2025-05-20:
This is not stale for me, as I don't think there's a way to execute a callback at the end of a timeline

lorenzo-arena, 2025-06-06:
Still not stale

AndreCostaaa, 2025-06-10:
Hi @lorenzo-arena 

I agree that it makes sense. Check #8403 with the solution

lorenzo-arena, 2025-06-10:
While that be a welcoming change for sure, I was mainly referring to animation timelines. As an example, consider this test I mostly copied from another timeline test:

```c
void test_anim_looping_timeline_without_exec_cb_but_anim_start_cb_and_completed_cb(void)
{
    lv_anim_t anim1;

    lv_anim_init(&anim1);
    lv_anim_set_values(&anim1, 0, 1024);
    lv_anim_set_duration(&anim1, 300);
    lv_anim_set_exec_cb(&anim1, anim1_exec_cb);
    lv_anim_set_start_cb(&anim1, anim1_start);
    lv_anim_set_completed_cb(&anim1, anim1_completed);
    anim1_start_called = 0;
    anim1_completed_called = 0;

    anim_timeline = lv_anim_timeline_create();
    lv_anim_timeline_add(anim_timeline, 200, &anim1);
    lv_anim_timeline_set_progress(anim_timeline, 0);
    lv_anim_timeline_start(anim_timeline);
    lv_anim_timeline_set_repeat_count(anim_timeline, 2);

    lv_refr_now(NULL);

    /*
     *           |----anim1----|
     *
     *   0       200   300    500
     */

    lv_test_wait(20); /*Wait 20 ms*/
    TEST_ASSERT_EQUAL(0, anim1_start_called);
    TEST_ASSERT_EQUAL(0, anim1_completed_called);

    lv_test_wait(200); /*Now we are at 220ms */
    TEST_ASSERT_EQUAL(1, anim1_start_called);
    TEST_ASSERT_EQUAL(0, anim1_completed_called);

    lv_test_wait(300); /*Now we are at 520ms */
    TEST_ASSERT_EQUAL(1, anim1_start_called);
    TEST_ASSERT_EQUAL(1, anim1_completed_called);

    lv_test_wait(200); /*Now we are at 720ms */
    TEST_ASSERT_EQUAL(2, anim1_start_called);
    TEST_ASSERT_EQUAL(1, anim1_completed_called);

    lv_test_wait(300); /*Now we are at 1020ms */
    TEST_ASSERT_EQUAL(2, anim1_start_called);
    TEST_ASSERT_EQUAL(2, anim1_completed_called);
}
```

This doesn't currently pass, even with the new patch, and I think is due to the fact that the following snippet

```c
if((!at->reverse && at->act_time == 0) || (at->reverse && at->act_time == playtime)) {
    for(uint32_t i = 0; i < at->anim_dsc_cnt; i++) {
        at->anim_dsc[i].is_started   = 0;
        at->anim_dsc[i].is_completed = 0;
    }
}
```

is only run once in `lv_anim_timeline_start`

AndreCostaaa, 2025-06-10:
Actually, the reason it's not passing is because you're calling `lv_anim_timeline_set_repeat_count` after `lv_anim_timeline_start`. Reversing the order of operations fixes it and the test passes