lvgl/lvgl #8222: Ninja is now available pre installed in the github actions images

kdschlosser, 2025-05-08:
### Introduce the problem

https://github.com/actions/runner-images/issues/11391

This will speed up compile times when using cmake. It works on Windows as well.

AndreCostaaa, 2025-05-09:
Hi, thank you, that's nice to know !
Tests are already built using ninja as you can see here:

https://github.com/lvgl/lvgl/blob/b1b66b78c2764267a320a1b12778747f280a39e0/tests/main.py#L136-L137

Is there another place where you'd like to see ninja being used ?

kdschlosser, 2025-05-09:
That is NOT using Ninja. That is telling cmake to generate the ninja scripts but I am not seeing where Ninja ever gets run. cmake doesn't run Ninja it will only generate the scripts that get used by Ninja when it runs. Ninja has to be run afterwards.

kdschlosser, 2025-05-09:
They also just added ninja to the CI images and I don't see anywhere in the CI scripts where Ninja gets installed.

NOPE it was never getting installed.

https://github.com/lvgl/lvgl/blob/b1b66b78c2764267a320a1b12778747f280a39e0/scripts/install-prerequisites.sh

kdschlosser, 2025-05-09:
scratch that last comment. I just saw it.. I must be blind..

Unless how cmake works is different on posix then it is on windows you have to run cmake and then run ninja to compile on Windows.

AndreCostaaa, 2025-05-09:
> Unless how cmake works is different on posix then it is on windows you have to run cmake and then run ninja to compile on Windows.

You can also let cmake handle it for you

Using `-GNinja` you're telling cmake to generate `Ninja` files

The line `cmake --build builddir` will invoke ninja under the hood for you

See the documentation about CMake Generators [here](https://cmake.org/cmake/help/latest/manual/cmake-generators.7.html)

AndreCostaaa, 2025-05-18:
As this has been stale for a week, I believe we can close this, feel free to comment if you believe that is not the case :)

closed 2025-05-18 by AndreCostaaa