lvgl/lvgl #8230: The text in the dropdown cannot be centered

IAMMX, 2025-05-10:
### LVGL version

master

### Platform

all

### What happened?

The text in the dropdown cannot be centered

[image]

### How to reproduce?

```c
void dropdown_test(void)
{
    lv_obj_t * dropdown = lv_dropdown_create(lv_screen_active());
    lv_obj_set_size(dropdown, 200, LV_SIZE_CONTENT);
    lv_obj_set_style_text_align(dropdown, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN); //This code didn't work
    lv_obj_center(dropdown);

    lv_dropdown_set_options(dropdown, "Option 1\nOption 2\nOption 3");
    lv_obj_t * dropdown_list=lv_dropdown_get_list(dropdown);
    lv_obj_set_style_text_align(dropdown_list, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN);
}
```

lizhaoming634, 2025-05-10:
Perhaps you can use ​lv_obj_set_style_pad_left(dd, 20, LV_PART_MAIN); ​ to adjust the left padding position.

IAMMX, 2025-05-13:
> Perhaps you can use ​lv_obj_set_style_pad_left(dd, 20, LV_PART_MAIN); ​ to adjust the left padding position.

This approach is just a temporary fix—the padding value has to change whenever the text length varies. @liamHowatt, could you take a look at this issue?

liamHowatt, 2025-05-13:
Currently, dropdowns automatically center the label if there is no symbol

```c
void dropdown_test(void)
{
    lv_obj_t * dropdown = lv_dropdown_create(lv_screen_active());
    lv_obj_set_size(dropdown, 200, LV_SIZE_CONTENT);
    lv_dropdown_set_symbol(dropdown, NULL);
    lv_obj_center(dropdown);

    lv_dropdown_set_options(dropdown, "Option 1\nOption 2\nOption 3");
    lv_obj_t * dropdown_list=lv_dropdown_get_list(dropdown);
    lv_obj_set_style_text_align(dropdown_list, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN);
}
```

[image]

https://github.com/lvgl/lvgl/blob/73925458fcac5b5b8969cd95b09b8d91273f877e/src/widgets/dropdown/lv_dropdown.c#L950-L962

IAMMX, 2025-05-14:
> Currently, dropdowns automatically center the label if there is no symbol

Is there a way to center the text while preserving the symbols?

liamHowatt, 2025-05-14:
I don't think so. Do you want it centered between the left side and the symbol or centered directly in the middle?

IAMMX, 2025-05-14:
> I don't think so. Do you want it centered between the left side and the symbol or centered directly in the middle?

Retain the symbols and center the font.

[image]

liamHowatt, 2025-05-14:
Ok so centered directly in the middle. Do you want to open a PR? If possible it'd be good to retain the previous behavior too.

IAMMX, 2025-05-15:
> Ok so centered directly in the middle. Do you want to open a PR? If possible it'd be good to retain the previous behavior too.

```c
void dropdown_test(void)
{
    lv_obj_t * dropdown = lv_dropdown_create(lv_screen_active());
    lv_obj_set_size(dropdown, 200, LV_SIZE_CONTENT);
    lv_obj_set_style_text_align(dropdown, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN); //This code didn't work
    lv_obj_center(dropdown);

    lv_dropdown_set_options(dropdown, "Option 1\nOption 2\nOption 3");
    lv_obj_t * dropdown_list=lv_dropdown_get_list(dropdown);
    lv_obj_set_style_text_align(dropdown_list, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN);
}
```

I'd be happy to open a PR, but I'm not sure how to modify it. The main issue is whether there's a way to make the code lv_obj_set_style_text_align(dropdown, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN); work.

liamHowatt, 2025-05-16:
I've started working on it. I just need to figure out why the calendar dropdowns are centering by default now.

https://github.com/lvgl/lvgl/compare/master...liamHowatt:lvgl:feat/dropdown-explicit-selected-align

IAMMX, 2025-05-17:
> I've started working on it. I just need to figure out why the calendar dropdowns are centering by default now.
> 
> [master...liamHowatt:lvgl:feat/dropdown-explicit-selected-align](https://github.com/lvgl/lvgl/compare/master...liamHowatt:lvgl:feat/dropdown-explicit-selected-align)

Thanks you

liamHowatt, 2025-06-03:
Sorry, I haven't taken time to come back to it. I will try next week.

IAMMX, 2025-07-04:
> Sorry, I haven't taken time to come back to it. I will try next week.

Hi, has there been any progress on this issue?😊

PR #8601 (2025-07-29): fix(dropdown): support aligning the dropdown button's text
Fixes #8230

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`ast [... cut]