lvgl/lvgl #8250: Garbled unl0kr graphics on Steam Deck

bjoern-tantau, 2025-05-15:
### LVGL version

v9.2.2

### Platform

OpenSUSE Tumbleweed running on a Steam Deck LCD.

### What happened?

I was advised in https://gitlab.postmarketos.org/postmarketOS/buffybox/-/issues/46#note_480280 to open a ticket here.

When trying to run unl0kr with the DRM backend on my Steam Deck I only get garbled output.

[image]

I'm also attaching the output of

```
# grep -r '.' /sys/class/graphics/*
# grep -r '.' /sys/class/vtconsole/*
# grep -r '.' /sys/class/drm/card*
```

[card.txt](https://github.com/user-attachments/files/20224483/card.txt)
[graphics.txt](https://github.com/user-attachments/files/20224485/graphics.txt)
[vtconsole.txt](https://github.com/user-attachments/files/20224484/vtconsole.txt)

bjoern-tantau, 2025-05-15:
unl0kr is compiled using https://build.opensuse.org/package/show/home:bjoern.tantau/buffybox

AndreCostaaa, 2025-05-18:
Hi @bjoern-tantau. Thank you for opening this issue, it's always nice to see where LVGL is getting used ! 

Would it be possible to attach the logs of the `unl0kr` application ?

H5117, 2025-05-22:
@bjoern-tantau You can change the string in `unl0kr/lv_conf.h` from
```
    #define LV_LOG_LEVEL LV_LOG_LEVEL_WARN
```
to
```
    #define LV_LOG_LEVEL LV_LOG_LEVEL_TRACE
```
, recompile the package, and attach the more verbose logs.

bjoern-tantau, 2025-05-24:
Here you go. The file is massive.

[unl0kr.log](https://github.com/user-attachments/files/20427092/unl0kr.log)

bjoern-tantau, 2025-05-25:
@michelealbrigo Maybe you can extract the binary with the verbose output from https://download.opensuse.org/repositories/home:/bjoern.tantau/openSUSE_Tumbleweed/x86_64/buffybox-unl0kr-567e303ag-23.1.x86_64.rpm and run that. You don't have to run it on boot. The buffyboard readme (or the unl0kr readme) contains instructions on running it outside of boot.

bjoern-tantau, 2025-05-31:
Don't worry about being late. You can't always control when you have time, energy and the desire to work on this. I'm disabled so it usually takes me a few days to have the energy to reply.

The log files above are from running unl0kr in a fully booted system.

The issue with it working in some kernel versions but not others was resolved. That was only relevant for FB and was simply because my distribution disabled the FB device in 6.12.9.

Before that I never used the DRM backend because it never worked on that device. The unl0kr folks updated LVGL to version 9.2.2 and with that I get at least the garbled output from the screenshot above. Before that I wouldn't get any output at all.

I don't know of any other DRM applications I could test. If you have a test application I could try that. I see the error in the running system so I wouldn't even have to put it into initramfs.

X11 and Gnome and Plasma Wayland all work without any issues.

hustlerone, 2025-06-09:
FYI this happens on the SD if you have the bootloader make the resolution be 1280x800

closed 2026-01-21 by bjoern-tantau