lvgl/lvgl #8258: you can add FreeType to online examples

bottle2, 2025-05-18:
### LVGL version

v9.3.0 dev

### Platform

your online demos

### What happened?

https://docs.lvgl.io/master/details/libs/freetype.html

it says "FreeType is not installed"

you can use the Emscripten option `--use-port=freetype` to add FreeType to a HTML5 build.

AndreCostaaa, 2025-05-18:
Hi @bottle2. Thank you for opening this issue ! 

Would you like to open a PR for it ?

bottle2, 2025-05-19:
I won't write a PR, maybe in 3 months if I graduate and remember about it. I also don't know CMake neither CI nor anything, so I feel uneasy to make changes. I'm just telling what I'm seeing as I learn your library for now.

regarding file system. Emscripten emulates a filesystem, all needed is flag `--embed-file=lvgl/examples/libs/freetype/Lato-Regular.ttf`. I dunno how you are shipping the online demos, so I don't know how it interacts with the flag `-sSINGLE_FILE`, but if you don't use this flag, the entire font file will be contained inside the generated `.wasm` file and normal C functions for reading files will work flawlessly.

more info:

https://emscripten.org/docs/porting/files/packaging_files.html#packaging-using-emcc
https://emscripten.org/docs/porting/files/file_systems_overview.html

it works like magic, my experiments so far worked in every first try

AndreCostaaa, 2025-05-19:
Ohh seems like a solved issue then. Thank you, let's see if someone wants to try to implement it!

closed 2025-06-16 by lvgl-bot