lvgl/lvgl #8267: Not able to flash the screen

keerthansys22, 2025-05-19:
### LVGL version

v9.2.0

### Platform

TI CCS IDE on linux

### What happened?

My objective is to flash the background screen (red and black), the flashing rate is 3 seconds on, 1 second off. I had implemented in my project, the issue I'm facing here is after 22 mins, the flashing stops and randomly screen is giving all kind of colors and starts to act weird. What might be the issue? I have added few code snippets too. Kindly help. Thanks.

### How to reproduce?

```static void flash_background_cb(lv_timer_t* timer) {
    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, screenmain);
    lv_anim_set_time(&a, 300); // Duration of the animation
    lv_anim_set_values(&a, 255, 0); // Fade out
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_obj_set_style_bg_opa);
    lv_anim_start(&a);

    // Change color after fade out
    if (is_red) {
        lv_obj_set_style_bg_color(screenmain, lv_color_black(), LV_PART_MAIN);
    } else {
        lv_obj_set_style_bg_color(screenmain, lv_color_hex(0xe80000), LV_PART_MAIN);
    }
    is_red = !is_red;

    // Fade in animation
    lv_anim_init(&a);
    lv_anim_set_var(&a, screenmain);
    lv_anim_set_time(&a, 300); // Duration of the animation
    lv_anim_set_values(&a, 0, 255); // Fade in
    lv_anim_set_exec_cb(&a, (lv_anim_exec_xcb_t)lv_obj_set_style_bg_opa);
    lv_anim_start(&a);
}

void disp_mod_status_screen(int status_value, int indx)
{
    screenmain = lv_obj_create(NULL);

    if(screenmain == NULL)
    {
        printf("Screen not created\r\n");
        return;
    }
    if ((screen_status_value[status_value].status) == 2)
    {
        if (flash_on == 1)
        {
            stop_flashing_timer(); // Stop any existing timer
        }

        set_background_color(screenmain, RED_BACKGROUND);

        status_label_creation(screenmain, "Device ID: ", 4, 4, 88, 16, &FAULT_FONT_SIZE_DEVICE_ID, WHITE_FONT);
        status_label_creation(screenmain, "Status: ", 4, 35, 68, 16, &FAULT_FONT_SIZE_STATUS, WHITE_FONT);
        status_label_creation(screenmain, "FAULT", 62, 35, 60, 16, &FAULT_FONT_SIZE_STATUS_VAL, WHITE_FONT);

        lv_obj_t* label3 = lv_label_create(screenmain);
        lv_label_set_long_mode(label3, LV_LABEL_LONG_SCROLL_CIRCULAR);     // Circular scroll
        lv_obj_set_size(label3, 120, 15);
        lv_label_set_text_static(label3, m_status.sts_status_desc);
        lv_obj_align(label3, LV_ALIGN_DEFAULT, 4, 50);
        lv_obj_set_style_text_font(label3, &FAULT_FONT_SIZE_STS_SCROLL, LV_PART_MAIN);
        lv_obj_set_style_text_color(label3, lv_color_white(), LV_PART_MAIN);

        // Load the screen
        lv_scr_load_anim(screenmain, LV_SCR_LOAD_ANIM_NONE, 0, 0, true);

        // Create a timer to toggle background color
        flash_timer = lv_timer_create(flash_background_cb, 600, NULL); // Adjusted duration for a faster flash
        flash_on = 1; // Set the flag to indicate flashing is active
    }
}

/**
 * @brief function to stop the flashing timer
 * @param none
 */
void stop_flashing_timer()
{
    if(flash_timer)
    {
        lv_timer_del(flash_timer); // Delete the timer
        flash_timer = NULL; // Reset the timer pointer
        flash_on = 0; // Reset the flashing state
    }
}
```

AndreCostaaa, 2025-05-19:
Hi @keerthansys22
22 minutes seems wildly specific. How many times has that happened ?

My first guess would be something overflowing somewhere if the error takes that much time to arise. Just to make sure, are you using v9.2.0 specifically and not latest master ?

AndreCostaaa, 2025-05-20:
Hi. Would you mind sharing your `lv_conf.h` ? I can take a look

keerthansys22, 2025-05-20:
This is the link to access - https://drive.google.com/file/d/15rtGTEFqwfFQLOiGzhtuPiinLqQqdLGq/view?usp=sharing

The changes I have done in lv_conf.h is
--> Changed the lv_color_depth 24
--> Increased LV_MEM_SIZE
--> Enabled QR library
--> Added some declaration for custom fonts

Thanks :)

keerthansys22, 2025-05-21:
Hi @AndreCostaaa, I have recorded the video of the screen bug that is happening after 22 minutes. Before this everything was working smoothly as expecting i.e. flashing of red and black screens with scrolling details. I have attached the screen bug recording.

https://github.com/user-attachments/assets/387ee9b6-3527-493e-9bee-9cbc3f7d7e6f
[image]

AndreCostaaa, 2025-05-21:
Oh, so you encounter this error in the target platform !

Have you ever encountered this error using the lvgl simulator ?
Could you create minimalist example that produces that error and share it with me ?
I suggest using the simulator with the SDL backend. The easiest way would be to produce your code using the [lv_port_pc_vscode](https://github.com/lvgl/lv_port_pc_vscode) repository

keerthansys22, 2025-05-21:
Sure, I will do this and update you. Thanks.

closed 2025-07-04 by lvgl-bot