lvgl/lvgl #8300: Slider Knob overshoots the endpoints by 1*Knob_Radius

FloHomi, 2025-05-26:
### LVGL version

v9.2

### Platform

Does not matter

### What happened?

It appears that the knob of the slider is in the wrong position. (half it's radius).  the problem is that a simple offeet in the one direction wouldn't do it. Then the knob is 2*R over on the other side.
It is hard to see, but I think in the online examples it is sligtly visible too.

[image]

### How to reproduce?

```
    lv_obj_t* const kSlider = lv_slider_create(kScreen);
    lv_obj_center(kSlider);
    lv_obj_set_style_pad_all(kSlider, -3, LV_PART_KNOB);
```

FloHomi, 2025-06-12:
not stale

liamHowatt, 2025-07-02:
Is it related to #8200?

closed 2025-07-26 by lvgl-bot