lvgl/lvgl #8321: Turns of the scale with numbers

brightproject, 2025-05-27:
### Introduce the problem

  I am creating such a scale on ESP32 with a 480x480 display

[image]

```
  ui_hdg_scale = lv_meter_create(approach_screen);

  lv_obj_set_pos(ui_hdg_scale, -158, 29);
  lv_obj_set_size(ui_hdg_scale, 800, 800);
  lv_obj_set_style_radius(ui_hdg_scale, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_text_font(ui_hdg_scale, &lv_font_montserrat_26, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_text_color(ui_hdg_scale, lv_color_hex(0xff0f710a), LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_bg_opa(ui_hdg_scale, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
  
  lv_meter_scale_t *scale = lv_meter_add_scale(ui_hdg_scale);
  lv_meter_set_scale_ticks(ui_hdg_scale, scale, 72, 3, 10, lv_color_hex(0xff0f710a));
  lv_meter_set_scale_major_ticks(ui_hdg_scale, scale, 6, 5, 10, lv_color_hex(0xff0f710a), 15);
  lv_meter_set_scale_range(ui_hdg_scale, scale, 0, 36, 355, 270);
  // lv_meter_set_scale_range(ui_hdg_scale, scale, 0, 36, 355, -(HDG - 270));
```
I need to turn the scale, if I start doing this I get this picture.
```
  // lv_meter_set_scale_range(ui_hdg_scale, scale, 0, 36, 355, 270);
  lv_meter_set_scale_range(ui_hdg_scale, scale, 0, 36, 355, -(HDG - 270));
```

[image]

https://github.com/user-attachments/assets/fe3f785e-be82-48c1-8937-fef201b5cf36

Tried to clear the display
`lv_obj_clean(ui_hdg_scale);`
But it doesn't help.

brightproject, 2025-06-15:
Actual

AndreCostaaa, 2025-06-16:
Thank you for sharing the solution <3