lvgl/lvgl #8357: Load screen execution time

Yash333code, 2025-06-04:
### Introduce the problem

I am using LVGL V 8.3.10  on esp32-S3 using EEZ studio. When i call "lv_scr_load(screen);"  or "lv_scr_load_anim(screen, LV_SCR_LOAD_ANIM_NONE, 0, 0, false);" it takes more than 600 msec to show changes on Screen. what are the possible cause of so and how can i take down this time to below 100 msec.

vwheeler63, 2025-06-19:
Hello, @Yash333code !  There are several factors that can be involved:

- Does the `lv_scr_load()` trigger a large number of objects/Widgets creation before the next display refresh can take place?
- Is there otherwise a long delay before the next display refresh takes place (normally done via `lv_timer_handler()`?
- If the delay really is due to a large number of Widgets being created on the new screen, consider pre-creating the screen and set of Widgets?

With a pre-created screen and Widget tree, the screen drawing of the new screen should begin almost immediately.

vwheeler63, 2025-06-27:
@kisvegabor  Thoughts?

AndreCostaaa, 2025-06-27:
What do you mean by `it says the FPS will be slow but is it this much slow.` ? Do you mean these warnings?

```bash
[Warn]	(0.000, +0)	lv_init: Memory integrity checks are enabled via LV_USE_ASSERT_MEM_INTEGRITY which makes LVGL much slower lv_init.c:316
[Warn]	(0.000, +0)	lv_init: Object sanity checks are enabled via LV_USE_ASSERT_OBJ which makes LVGL much slower lv_init.c:320
[Warn]	(0.000, +0)	lv_init: Style sanity checks are enabled that uses more RAM lv_init.c:324
```

Also have you tried compiling in release mode ?

closed 2025-07-21 by lvgl-bot