lvgl/lvgl #8358: Problem setting up DMA2D accelerator with LV_USE_DRAW_DMA2D?

LuisCRSousa, 2025-06-04:
### LVGL version

v9.3.0

### Platform

STM32H7, 16b color depth with freertos 

### What happened?

Just upgraded from 9.2.2 to 9.3.0 to use DMA2D with LV_USE_DRAW_DMA2D but i'm having problems and questions setting up the DMA2D.

My display and ltdc driver are being created with `lv_st_ltdc_create_direct` with dual buffers and it works without dma2d but not with dma2d active.
I also tried with and without LV_USE_DRAW_DMA2D_INTERRUPT but it also does not work.

Do I need to activate DMA2D in the cubeMX? It does not work with or without.

Documentation does not clarify what is the procedure when enabling LV_USE_DRAW_DMA2D.

LuisCRSousa, 2025-06-06:
> What kind of error you are having? In general enabling the DMA2D on the LVGL is sufficient to get it running, since the draw unit configures everything.
> 
> Does the flush does not occur? It crashes?

It is being asserted at lv_tlsf.c
Still debugging it

`size_t lv_tlsf_free(lv_tlsf_t tlsf, const void * ptr)
{
    size_t size = 0;
    /* Don't attempt to free a NULL pointer. */
    if(ptr) {
        control_t * control = tlsf_cast(control_t *, tlsf);
        block_header_t * block = block_from_ptr(ptr);
        tlsf_assert(!block_is_free(block) && "block already marked as free");
        size = block->size;
        block_mark_as_free(block);
        block = block_merge_prev(control, block);
        block = block_merge_next(control, block);
        block_insert(control, block);
    }

    return size;
}`

joyplayer, 2025-06-09:
same to me

uLipe, 2025-06-10:
Could you post your lv_conf.h?

I'm suspecting you are facing alignment issues, in general DMA2D has a strict alignment requirement for the buffers to being aligned to the cache line

closed 2025-07-03 by lvgl-bot